serialport/node-serialport #1078: Data loss on input

RadoslavMarinov, 2017-01-24:
SerialPort version: 4.0.5
NodeJS Version: v6.9.2
Operating System and Hardware Platform: Intel i5-6300
Are you using an alternative NodeJS runtime? (eg Electron): No

### Summary of Problem
Misses data on the serial input. 
### Steps and Code to Reproduce the Issue

var buffer = new Buffer('');

var port = new SerialPort("COM9", {
  baudRate: 115200,
  rtscts: true
});

port.on('data', function (data) {
	buffer = Buffer.concat([buffer, data]);
});

reconbot, 2017-02-13:
Hey sorry it took so long to respond to this. Can you give some more details?

RadoslavMarinov, 2017-02-23:
I am sniffing the same data at the same time  via two tools (on different ports of course)

The one is: NodeJS program using serial port module
The other is [Hercules serial port terminal](http://www.softpedia.com/get/Network-Tools/Telnet-SSH-Clients/Hercules-SETUP.shtml).

Then I compare  the two log files using program called [Winmerge](http://winmerge.org/)

Both files resemble almost exactly, except in some places the log file produced by the Hercules, has more bytes then the NodeJS program. 

When the two log files are passed to a parser, the one from the Hercules reports no BAD_CRC(data integrity check) and the one from the NodeJS reports BAD_CRC exactly on the places where the data differs.

RadoslavMarinov, 2017-02-23:
Oh sorry I found the mistake. This `console.log(data)` screws everything. It looks like it takes too much time and the new data is missed. When I removed that line everything works just fine.

reconbot, 2017-03-03:
I couldn't tell you the behavior of two programs reading the same serialport, there's a good chance they'll interfere with eachother.

reconbot, 2017-03-03:
Glad it worked out!

closed 2017-03-03 by reconbot