serialport/node-serialport #1079: Error reading/writing

nivoset, 2017-01-27:
SerialPort version: 4.0.7
NodeJS Version: 7.4.0
Operating System and Hardware Platform: Raspberry Pi3 raspian
Are you using an alternative NodeJS runtime? (eg Electron): Node

### Summary of Problem
I am having some communication problems with serialport on Pi3 (was working fine on Pi2) for testing i hooked up a USB to serial adapter looped back to the ttyS0 input. and im trying to ping pong a message 'test\r\n' back and forth. i'm using the readline parser ('\r\n') and it works... sometimes, but most of the time i get garbled garbage 'tetest' or other random messed up garbage for most items it sends. i have gone though and even put in core_freq=250 on the pi3 to see if its a problem with it.

code below is ugly, light module just does some flashing lights i was going to use to verify it all.

### Steps and Code to Reproduce the Issue

var light = require('./light.js');
var serialPort = require("serialport");

var BAUD = 19200;
var PARSER = serialPort.parsers.readline("\r\n");
var failed = {USB: 0, AMA: 0};
//initial power up to show led is working
light.red.blink({onTime: 4000, count: 1});
//light.green.blink({onTime: 5000, count: 1});
//set green to heartbeat mode.
light.green.blink({});

var deviceList = [
   {      dev: "/dev/ttyS0",      isCon: false,      conError: 3,      recError: 5  ,failureCount: 0, message: 'Test', failed: 0},
   {      dev:"/dev/ttyUSB0",     isCon: false,      conError: 2,      recError: 4  ,failureCount: 0, message: 'Test', failed: 0}];
var dev = [];

var checkMessage = function (device, message) {
   var msg = deviceList[((device == "/dev/ttyS0")?1:0)];
   var dev = deviceList[((device == "/dev/ttyS0")?0:1)];
   if (message == msg.message) {
      dev.failureCount = 0;
      msg.failureCount ++;
   } else {
      //console.log(device + " \t " + message + " \t " + msg.message);
      dev.failureCount ++;
   }
};

var checkFailures = function () {
   if (deviceList[0].failed == 0 && deviceList[1].failed == 0) {
      for (var i in deviceList) {
         if (deviceList[i].failureCount > 50) {
            deviceList[i].failed = 1;
            console.log(deviceList[i].dev + " Failed to Recieve/Transmit");
            light.red.blink({repeatDelay:1000,count:deviceList[i].recError});
         }
      }
   }
   setTimeout(checkFailures, 1000);
};
setTimeout(checkFailures, 100);

var newCon = function (device, faultID) {
   console.log(device.dev + " is defined");
   var self = this;
   self.device = device.dev;
   
   var dev = new serialPort(device.dev, {
      baudrate: BAUD,
      parser: PARSER,
      //lock: false
   }, function (err) {
      if (err) {
         if (deviceList[0].failed == 0 && deviceList[1].failed == 0) {
            console.log("Failed to open Device: " +  device.dev);
            console.log("Error: " + err);
            light.red.blink({onTime:500, offTime:500, repeatDelay:1000,count:deviceList[1].conError});
            device.failed = 1;
         }
      } else {
         console.log(device.dev + " is connected... ");
         dev.on('data', function(data) {
            //if (device.dev == deviceList[1].dev)  
            console.log(device.dev + ' received: ' + data + " " + (data == "Test"));
            checkMessage(device.dev, data);
         });
         setTimeout(sendMessage, 1500);
      }
   });
   

   var sendMessage = function() {
      dev.write(device.message.toString()+"\r\n");
      //dev.write("test\r\n");
      device.failureCount++;
      setTimeout(sendMessage, 1000); 
   };

};

setTimeout(function() {
   //connect devices
   for (var i in deviceList) {
      dev.push(new newCon(deviceList[i]));
   }
},5000);

i get this as output:
/dev/ttyS0 received:
TTestes false
/dev/ttyUSB0 received: t false
/dev/ttyS0 received: TeTstes false
/dev/ttyUSB0 received: t false
/dev/ttyS0 received: TestTest false
/dev/ttyUSB0 received:  false
/dev/ttyS0 received: TTeste false
/dev/ttyUSB0 received: st false
/dev/ttyS0 received: TesTest false
/dev/ttyUSB0 received: t false
/dev/ttyS0 received: TTestes false
/dev/ttyUSB0 received: t false
/dev/ttyS0 received: TesTestt false
/dev/ttyUSB0 received:  false
stev/ttyUSB0 received: TesTet false

nivoset, 2017-01-27:
so i find if i offset them, it works fine (they alternate sending every half second or so) then everything seems to work fantastic with no garbling.

fivdi, 2017-01-28:
Have shell and kernel messages on the serial connection been disabled?

With Raspbian they can be disabled by running `sudo raspi-config`, navigating to "7 Advanced Options / A8 Serial" and answering the question "Would you like a login shell to be accessible over serial?" with "No".

nivoset, 2017-01-28:
it should be turned off, i will double check again,

but what i do see is bleed over from the other comm line, not the console.
if i do one side send 1234, the other 5678, i will see messages like
12563478, both messages put together, if i slow down the communication to
not happen at the same time, it works fine

closed 2017-03-02 by reconbot