serialport/node-serialport #1083: Uncaught Error: Cannot find module 'serialport' after packaging electron app (electron-packager)

jakoguta, 2017-02-05:
SerialPort version: 4.0.7

NodeJS Version: 6.9.4

Operating System and Hardware Platform: Windows 10 x64, intel core i3-2310M CPU @ 2.10GHz, 4GB RAM

Are you using an alternative NodeJS runtime? (eg Electron): Yes, am using electron

### Summary of Problem

Hi, our app uses the ``` node-serialport ``` package. Works fine in development, but on packaging using ``` electron-packager ``` 
it fails with the error:

```
module.js:457 Uncaught Error: Cannot find module 'serialport'
```

### Steps and Code to Reproduce the Issue

* Download starter app [angular2-electron] (https://github.com/joaogarin/angular2-electron). Edit the login page so that you
you can go directly to home page.
* Install node-serialport ``` npm install serialport --save ```
* In home.component.ts, initialize the serialport in the import section:
```
...
import { Location } from "@angular/common";

declare var require: any
var SerialPort = require('serialport');

...
```
* Package the electron app using electron-packager by running the following command ``` npm run package ```
* Go to release and run the app.

N/B: I have even tried modifing the ``` package.json ``` script section to:

```
"scripts": {
    ...
    "package-win": "electron-packager src/app --overwrite --asar=true --platform=win32 --arch=x64 --icon=src/app/dist/assets/app-icon --prune=true --out=release-builds"
  }
```
and 
```
"scripts": {
    ...
    "package-win": "electron-packager src/app --overwrite --asar.unpack=\"node_modules/serialport/build/**\" --asar=true --platform=win32 --arch=x64 --icon=src/app/dist/assets/app-icon --prune=true --out=release-builds"
  }
```
I have tried modifing ``` --asar.unpack ``` to various paths ``` --asar.unpack=\"*.node\" ``` but still not packing the node-serialport. Any pointers on 
how to package the app using ``` electron-packager ``` and have it run in production. Am yet to make the installer.

ferrisimo, 2017-02-05:
I ended up using electron-builder to create a .exe installer which works OK. What did cause issue was that serialport needed to be build to match electron. ie npm install serialport works with the node cli OK but not packaged in electron.

In my setup I have "serialport": "^4.0.6" in package.json and I've rebuilt serialport against version 1.4.15 of electron by doing this......

cd ./node_modules/serialport/
node-gyp rebuild --target=1.4.15 --dist-url=https://atom.io/download/atom-shell

Run your packager after that and I'd expect it to work.
Hope this goes some way to help.

jakoguta, 2017-02-06:
Let me try out this and get back to you.

jakoguta, 2017-02-06:
@ferrisimo  I am having trouble with my ```npm run package```.  The ```serialport.node``` path is not anywhere near a production friendly path i.e. 

```
bindings.js:91 Uncaught Error: Could not locate the bindings file. Tried:
 → D:\<dev-path>\projectName\build\serialport.node
 → D:\<dev-path>\projectName\build\Debug\serialport.node
 → D:\<dev-path>\projectName\build\Release\serialport.node
 → D:\<dev-path>\projectName\out\Debug\serialport.node
 → D:\<dev-path>\projectName\Debug\serialport.node
 → D:\<dev-path>\projectName\out\Release\serialport.node
 → D:\<dev-path>\projectName\Release\serialport.node
 → D:\<dev-path>\projectName\build\default\serialport.node
 → D:\<dev-path>\projectName\compiled\6.5.0\win32\x64\serialport.node
```
Also am having a warning from webpack when I ```npm run build``` am not sure if this should be cause for alarm : 

``` 
WARNING in ./~/bindings/bindings.js
76:43 Critical dependency: the request of a dependency is an expression
```

I would like to switch to ```electron-builder``` like you said you did. Please would you share your modifications in the following files ```package.json``` and ```webpack.config.js``` that allowed you to use ```electron-builder```. Am new to this and my trials are proving to be a mess.

ferrisimo, 2017-02-06:
Essentially all you need to do is npm install -g electron-builder

Then add to your scripts section....

"scripts": {
    "pack": "build --dir",
    "dist": "build"
  },

Then a new section below the dependencies which identifies your icons for each target eg

  "dependencies": {
  ...
  "serialport": "^4.0.6"
  },
  "build": {
    "appId": "Your App Id",
    "mac": {
      "icon": "./app/img/app-icon.icns",
      "category": "Whatever"
    },
    "win": {
      "icon": "./app/img/app-icon.ico",
      "iconUrl": "./app/img/app-icon.ico"
    }
  }

jakoguta, 2017-02-09:
@ferrisimo I managed to get it working with ```electron-builder```. Thank you for the help. This issue may now be closed.

closed 2017-02-13 by reconbot