serialport/node-serialport #1084: OSX returns ports that cannot be opened by default

reconbot, 2017-02-07:
SerialPort version: 5.x
Operating System and Hardware Platform: OSX

### Summary of Problem
OSX `SerialPort.list` returns `/dev/cu.xxx` ports but tries to lock them by default and they cannot be locked. Only `/dev/tty.xxx` ports can be locked.

Thanks to #1070 and @lynndylanhurley for the report!

dcyoung, 2017-05-02:
I saw the comments about passing `{lock:false}` into the open options, but is there any workaround that permits the use of the `SerialPort.list`  

If not, is there a way to catch or avoid this error when using `SerialPort.list`, even if we can't properly return the port?

reconbot, 2017-07-17:
@dcyoung this issue is requesting someone make serialport list do what you ask

reconbot, 2017-07-17:
I swear to god there doesn't seem to be a c++ way to do this. I'm having a hell of a time finding docs on the functions we use. Maybe a string replace from `/dev/cu.` to `/dev/tty.` is in order, unless anyone has better ideas. =|

closed 2017-07-19 by reconbot

reconbot, 2017-07-19:
Figured it out, will have the tty ports in the next release.

lynndylanhurley, 2017-07-19:
Fantastic work @reconbot :smiley:

PR #1242 (2017-07-19): [osx] Return the tty instead of the cu for osx serialports
For the record, tty's are `DialinDevice`s and cu's are `CalloutDevice`s

Thanks to @kishinmanglani for helping me solve this!

The closest we could come to docs. https://opensource.apple.com/source/IOSerialFamily/IOSerialFamily-7/IOSerialFamily.kmodproj/IOSerialKeys.h

closes #1084