serialport/node-serialport #1085: On slackware 14.2, udevadm isn't on the default path

jmarca, 2017-02-07:
SerialPort version: tested on ^4.0.6, 5.0.0-beta3
NodeJS Version: v7.2.0
Operating System and Hardware Platform: slackware 14.2 (-current)
Are you using an alternative NodeJS runtime? (eg Electron): no

### Summary of Problem

udevadm is not on the default path for non-privileged users.  /sbin/ and /usr/sbin/ are only on super user path.

### Steps and Code to Reproduce the Issue

remove /sbin from path, udevadm is not discoverable.

The error I see is:

```
james@emma serialport_tests$ rm ~/bin/udevadm 
james@emma serialport_tests$ ./node_modules/.bin/serialport-list 
{ Error: Command failed: udevadm info --query=property -p $(udevadm info -q path -n /dev/ttyS1)
/bin/sh: udevadm: command not found
/bin/sh: udevadm: command not found

    at ChildProcess.exithandler (child_process.js:211:12)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at maybeClose (internal/child_process.js:885:16)
    at Socket.<anonymous> (internal/child_process.js:334:11)
    at emitOne (events.js:96:13)
    at Socket.emit (events.js:188:7)
    at Pipe._handle.close [as _onclose] (net.js:501:12)
  killed: false,
  code: 127,
  signal: null,
  cmd: 'udevadm info --query=property -p $(udevadm info -q path -n /dev/ttyS1)' }
```

I made a note on the commit in which this changed earlier.

Slackware tends to be a bit old-fashioned/traditional about these things, and most people who run Slackware will know how to fix this problem (just symlink the /sbin/udevadm command to ~/bin or something).  Given that this is software that runs on all kinds of Linux flavors, including ones in which udevadm is stashed under /bin not /sbin, I can't see a better solution.  

Perhaps a try catch loop, if it fails, try sbin, then give up?  I just hacked up something ugly, not at all in your coding style:

```
const exec = require('child_process').exec;
var file = 'tty0'
const udevadm = `udevadm info --query=property -p $(udevadm info -q path -n ${file})`;
const sbin_udevadm = `/sbin/udevadm info --query=property -p $(/sbin/udevadm info -q path -n ${file})`;

exec(udevadm, (error, stdout, stderr) => {
  if (error) {
      console.error(`exec error: ${error}`);
      exec(sbin_udevadm,(e2,stdout2,stderr2) => {
          if(e2){
              console.error(`exec error: ${e2}`);
              return;
          }
          console.log(`stdout2: ${stdout2}`);
          console.log(`stderr2: ${stderr2}`);
          return
      })
  }
  console.log(`stdout: ${stdout}`);
  console.log(`stderr: ${stderr}`);
  return
});
```

Nested error handling is pretty inelegant, but it works on my machine and is a start perhaps?

```
james@emma serialport_tests$ node list.js 
exec error: Error: Command failed: udevadm info --query=property -p $(udevadm info -q path -n tty0)
/bin/sh: udevadm: command not found
/bin/sh: udevadm: command not found

stdout: 
stderr: /bin/sh: udevadm: command not found
/bin/sh: udevadm: command not found

stdout2: DEVNAME=/dev/tty0
DEVPATH=/devices/virtual/tty/tty0
ID_MM_CANDIDATE=1
MAJOR=4
MINOR=0
SUBSYSTEM=tty
USEC_INITIALIZED=6555318

stderr2: 
james@emma serialport_tests$ 
```

If the first error wasn't so chatty, I'd never even know it needed to find /sbin/ version of udevadm

(I tried a hack with `which udevadm`, but that would error out anyway with "no udevadm found in paths ....blah blah blah....  Seemed easier just to try, if it works, great, if it fails, try the most likely alternate).

reconbot, 2017-02-26:
Well... I'm a bit torn, I think it should be a clear requirement for serialport in the docs that `linux.list` needs `udevadm` on the path to operate. I'm very hesitant to hard code paths but I think we can error on the side of inclusivity.

jmarca, 2017-02-26:
I agree with you.  I also hate hard coding paths.  And doing something
like a call to `find` would be expensive.

I think adding a note to the docs is a perfectly good solution.  That
would have worked for me, because I read the README even *before* I
ran into troubles.

reconbot, 2017-02-26:
> node-serialport/issues/1085#issuecomment-282526453

Happy to land if you want to add udevadm to the requirements for linux, and
maybe a section about the error with a note about slackware. (Just make
sure to edit the template not the README.md)

---
Francis Gulotta
wizard@roborooter.com

reconbot, 2017-06-20:
I'm going to close this issue due to it's age, but if you'd like to continue with it feel free to comment and we'll reopen.

closed 2017-06-20 by reconbot

jmarca, 2017-06-20:
Yeah. Close it. Sorry I totally forgot to do anything on this. If I get time to doc properly I will and will submit a pull request.

reconbot, 2017-06-20:
Thanks

---
Francis Gulotta
wizard@roborooter.com