serialport/node-serialport #1086: Can't install on Win7 32 bit platform

delliottg, 2017-02-07:
SerialPort version: 4.0.7
NodeJS Version: 6.9.5
Operating System and Hardware Platform: Windows 7 Pro 32 bit
Are you using an alternative NodeJS runtime? (eg Electron):  No, downloaded directly from Node.JS site

### Summary of Problem

Please see [StackOverflow question here](http://stackoverflow.com/questions/42100458/cant-install-node-serialport-on-german-win7).  I am no longer convinced that the fact that it's German Windows is a factor.  This install is being run remotely on a machine in Germany via VNC Viewer Plus from my dev box in the Seattle, Washington area.

Errors received when trying to install serialport via npm (I can attach the error npm-debug.log if wanted, but it's nearly 15k lines):

	C:\Program Files\Sea-Bird\sbSerialWidget>npm install

	> serialport@4.0.7 install C:\Program Files\Sea-Bird\sbSerialWidget\node_modules\serialport
	> node-pre-gyp install --fallback-to-build

	node-pre-gyp ERR! Tried to download(undefined): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v48-win32-ia32.tar.gz
	node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and node@6.9.5 (node-v48 ABI) (falling back to source compile with node-gyp)

	C:\Program Files\Sea-Bird\sbSerialWidget\node_modules\serialport>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" clean )  else (node "" clean)

	C:\Program Files\Sea-Bird\sbSerialWidget\node_modules\serialport>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" configure --fallback-to-build "--module=C:\Program Files\Sea-Bird\sbSerialWidget\node_modules\serialport\build\Release\serialport.node" --module_name=serialport "--module_path=C:\Program Files\Sea-Bird\sbSerialWidget\node_modules\serialport\build\Release" --python=C:\Users\SeabirdUSA\.windows-build-tools\python27\python.exe --msvs_version=2015 )else (node "" configure --fallback-to-build"--module=C:\Program Files\Sea-Bird\sbSerialWidget\node_modules\serialport\build\Release\serialport.node" --module_name=serialport "--module_path=C:\Program Files\Sea-Bird\sbSerialWidget\node_modules\serialport\build\Release" --python=C:\Users\SeabirdUSA\.windows-build-tools\python27\python.exe --msvs_version=2015 )
	gyp WARN install got an error, rolling back install
	gyp ERR! configure error
	gyp ERR! stack Error: self signed certificate in certificate chain
	gyp ERR! stack     at Error (native)
	gyp ERR! stack     at TLSSocket.<anonymous> (_tls_wrap.js:1079:38)
	gyp ERR! stack     at emitNone (events.js:86:13)
	gyp ERR! stack     at TLSSocket.emit (events.js:185:7)
	gyp ERR! stack     at TLSSocket._finishInit (_tls_wrap.js:603:8)
	gyp ERR! stack     at TLSWrap.ssl.onhandshakedone (_tls_wrap.js:433:38)
	gyp ERR! System Windows_NT 6.1.7601
	gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "configure" "--fallback-to-build" "--module=C:\\Program Files\\Sea-Bird\\sbSerialWidget\\node_modules\\serialport\\build\\Release\\serialport.node" "--module_name=serialport" "--module_path=C:\\Program Files\\Sea-Bird\\sbSerialWidget\\node_modules\\serialport\\build\\Release" "--python=C:\\Users\\SeabirdUSA\\.windows-build-tools\\python27\\python.exe" "--msvs_version=2015"
	gyp ERR! cwd C:\Program Files\Sea-Bird\sbSerialWidget\node_modules\serialport
	gyp ERR! node -v v6.9.5
	gyp ERR! node-gyp -v v3.4.0
	gyp ERR! not ok
	node-pre-gyp ERR! build error
	node-pre-gyp ERR! stack Error: Failed to execute 'node-gyp.cmd configure --fallback-to-build --module=C:\Program Files\Sea-Bird\sbSerialWidget\node_modules\serialport\build\Release\serialport.node --module_name=serialport --module_path=C:\Program Files\Sea-Bird\sbSerialWidget\node_modules\serialport\build\Release --python=C:\Users\SeabirdUSA\.windows-build-tools\python27\python.exe --msvs_version=2015' (1)
	node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (C:\Program Files\Sea-Bird\sbSerialWidget\node_modules\serialport\node_modules\node-pre-gyp\lib\util\com
	pile.js:83:29)
	node-pre-gyp ERR! stack     at emitTwo (events.js:106:13)
	node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
	node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:877:16)
	node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
	node-pre-gyp ERR! System Windows_NT 6.1.7601
	node-pre-gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\Sea-Bird\\sbSerialWidget\\node_modules\\serialport\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp" "install" "--fallback-to-build"
	node-pre-gyp ERR! cwd C:\Program Files\Sea-Bird\sbSerialWidget\node_modules\serialport
	node-pre-gyp ERR! node -v v6.9.5
	node-pre-gyp ERR! node-pre-gyp -v v0.6.32
	node-pre-gyp ERR! not ok
	Failed to execute 'node-gyp.cmd configure --fallback-to-build --module=C:\Program Files\Sea-Bird\sbSerialWidget\node_modules\serialport\build\Release\serialport.node --module_name=serialport --module_path=C:\Program Files\Sea-Bird\sbSerialWidget\node_modules\serialport\build\Release --python=C:\Users\SeabirdUSA\.windows-build-tools\python27\python.exe --msvs_version=2015' (1)
	npm WARN sbeserialwidget@0.0.1 No description
	npm WARN sbeserialwidget@0.0.1 No repository field.
	npm ERR! Windows_NT 6.1.7601
	npm ERR! argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "install"
	npm ERR! node v6.9.5
	npm ERR! npm  v3.10.10
	npm ERR! code ELIFECYCLE

	npm ERR! serialport@4.0.7 install: `node-pre-gyp install --fallback-to-build`
	npm ERR! Exit status 1
	npm ERR!
	npm ERR! Failed at the serialport@4.0.7 install script 'node-pre-gyp install --fallback-to-build'.
	npm ERR! Make sure you have the latest version of node.js and npm installed.
	npm ERR! If you do, this is most likely a problem with the serialport package,
	npm ERR! not with npm itself.
	npm ERR! Tell the author that this fails on your system:
	npm ERR!     node-pre-gyp install --fallback-to-build
	npm ERR! You can get information on how to open an issue for this project with:
	npm ERR!     npm bugs serialport
	npm ERR! Or if that isn't available, you can get their info via:
	npm ERR!     npm owner ls serialport
	npm ERR! There is likely additional logging output above.

	npm ERR! Please include the following file with any support request:
	npm ERR!     C:\Program Files\Sea-Bird\sbSerialWidget\npm-debug.log

### Steps and Code to Reproduce the Issue

I've successfully installed:

 - Node.js 6.9.5

 - NPM 3.10.10

 - Python 2.7.13

 - Windows build tools from the instructions [found here][1].

 - Visual Studio build tools for VS 2015 (also tried with VS 2013)

 - Node gyp

 - Node pre-gyp

Then I tried making these [configuration change][2]s:

    $ npm config set msvs_version 2015
    $ set VCTargetsPath=C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140

No change to the error.

Here's our package.json:

	{
	  "name": "sbeserialwidget",
	  "version": "0.0.1",
	  "description": "",
	  "main": "server.js",
	  "directories": {
		"test": "test"
	  },
	  "scripts": {
		"test": "node_modules\\.bin\\mocha --reporter spec"
	  },
	  "license": "ISC",
	  "dependencies": {
		"chai": "^3.5.0",
		"dns": "^0.2.2",
		"express": "^4.14.0",
		"fs-extra": "^0.30.0",
		"mocha": "^3.0.2",
		"moment": "^2.15.1",
		"os": "^0.1.1",
		"request": "^2.74.0",
		"serialport": "^4.0.1",
		"socket.io": "^1.4.8",
		"username": "^2.2.2"
	  }
	}

reconbot, 2017-02-13:
Hrm...

Have you tried to [create a c++ project in visual studio](https://github.com/EmergingTechnologyAdvisors/node-serialport#windows)? Visual studio doesn't always set everything up until you do that.

delliottg, 2017-02-13:
This is on a client machine that doesn't actually have VS installed.  Would a copy of VS Express work?

Our IT department is getting me a copy of German Win7 so I can build a VM with it to test with.  We're wondering if the failure to build is because of something like a "falsch" being returned instead of "false", or a comma for a decimal point due to the German build of Windows.

reconbot, 2017-02-13:
I have no idea if that would work. @noopkat or @munyirik might however?

It's weird to me that the the binary didn't download, we have a prebuilt binary for that platform. You're probably going to have more luck going down that debugging path.

You can download the binary from [here](https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v48-win32-ia32.tar.gz) and put it in `node-serialport/build/Release/serialport.node`

munyirik, 2017-02-13:
From that log it looks like you ran into [this issue](https://github.com/nodejs/node-gyp/issues/695)

delliottg, 2017-02-15:
Today I was able to get back to this.  I created a Win7-32 VM, did an install of our software, then tested it to make sure it worked as desired.  I copied it from the VM to a server the German machines could see, zipped it up, copied it to the German machine and unzipped it in place (after deleting the entire previous install).  No go, so I uninstalled Node.js & reinstalled from the internet.  Copied the zipped folder into place again, still no go, so I deleted the \node_modules directory from our install and ran `npm install` on our `package.json` & voila, it worked (sort of, IE doesn't like it due to cross origin resource sharing issues, but it works fine with Chrome).

reconbot, 2017-02-17:
Great!

---
Francis Gulotta
wizard@roborooter.com

delliottg, 2017-02-17:
Yesterday I installed our software on a second German Win7 32 machine using what I learned the first time.  While the install still took over 2 hours of fiddling, at least I was able to get it installed.  A short list of what's required is (from the wiki article I wrote for the issue):

When installing the Service Check In software in Germany, we ran into multiple problems with the install.  Here's a summary of what I've done to fix the problems.

  - Install appropriate version of [https://nodejs.org/en/download/ Node.js] from the internet.  In the case of the German machines, they're 32 bit Windows machines.

  - [http://stackoverflow.com/questions/9626990/receiving-error-error-ssl-error-self-signed-cert-in-chain-while-using-npm Update npm on the machine]:

    // Do this first, or the upgrade will fail

      `npm config set ca ""`

      `npm install npm -g`

    // Undo the previous config change

      `npm config delete ca`
 You're quite likely to run into a SELF_SIGNED_CERT_IN_CHAIN error, if so, dig through the linked Stack Overflow article.  I've found that running `npm config set strict-ssl false -g` while a bad idea, makes this particular problem go away.  If it doesn't, try some of the other things suggested in that thread.  Not overly helpful I know, but this is fiddly and there doesn't seem to be a clear path through the various problems you're going to encounter.

  - [https://github.com/nodejs/node-gyp Install node-gyp along with Python 2.7x & Windows build tools] Follow the instructions for Windows, the first option for Windows has worked thus far.  Be patient, this takes some time to build & compile.

  -  [https://www.npmjs.com/package/node-pre-gyp Install node-pre-gyp] using the instructions that use `npm`.

  -  Copy the Service Check In bits from: `[redacted]`, to `[redacted].  You can ignore the /node_modules directory, you're going to recreate it locally with `npm install`

  - Navigate to the \sbserialwidget\ directory and try running `npm install`.  If all goes well, it'll install and you won't run into problems, but don't hold your breath.  If it seems to go well, try running `node server.js`.  If it works, basically nothing happens, the prompt just moves down a line.  If it fails it'll tell you which module it failed on (most likely `express` or `socket.io` at least in my experience, assuming that you've gotten the serialport library sorted out above).  If that happens go to the next bullet point

  - If the last command fails '''and you're doing a 32 bit install''''''', delete the \node-modules directory from the \sbserialwidget\ directory, then go copy & unzip `[redacted server location of a 32 bit install]` somewhere locally on the machine, then copy the files to the \sbserialwidget\ directory including the \node-modules\ directory.  Assuming everything went well, running `node server.js` should now give you a working node server.  

We haven't had a 64 bit install fail that we're aware of, but I'll update this if we do and figure it out.

reconbot, 2017-02-17:
>    error-error-ssl-error-self-signed-cert-in-chain-while-using-npm

What's "sbserialwidget"?

---
Francis Gulotta
wizard@roborooter.com

delliottg, 2017-02-17:
We're an oceanographic instrument manufacturer named Sea-Bird (EG: the "drone" the Chinese navy stole a few weeks back was packed with a bunch of our instruments).  Nearly all of our instruments talk over serial because we can push the signal literally miles up from the ocean floor using that protocol.  **SBSerialWidget** is the little Node.js server we've written that allows us to talk to a client machine's serial ports via a remotely served PHP website with a MySQL database attached for storing data, tests, configurations, etc.  It's mostly used by our Service department here in the Seattle area, but our German Service department also wants to use it, hence my installs over there.  We're only on our 4th or 5th sprint, but it's coming along nicely, and it wouldn't have been possible without your serialport library.

reconbot, 2017-06-20:
This is amazing, thank you for sharing. 

A few notes.

-5x-beta is coming along its proving to be a little more stable in the ways 4x wasn't but is still shaking out it's own issues.
- I'm dropping support for node 4 32bit because of similar CA issues.

I'm going to close this issue due to it's age, but if there's anything you'd like to continue with it feel free to comment and we'll reopen.

closed 2017-06-20 by reconbot

delliottg, 2017-06-20:
Does this mean you've dropped 32 bit completely, or it'll be in the -5x-beta?  Will it require moving our version of Node to 6.xxx?  I haven't had time to get back to the German Win7 32 bit problem.  They are currently able to use the software, but I've had to disable our auto-updates as they don't work properly there.  We believe this is because of a 200-300ms latency delay in our ping times between there and our PHP / MariaDB server, probably due to the VPN software we use.  Supposedly our parent company is working on that pipeline and things should speed up, but we don't know when that should be.

Just as kudos to you and your team, our service department is using our software (_which is built on your software_) on a daily basis, which has improved their throughput by roughly 25-30% (we only have estimated numbers so far).  Since their check-in & check-out scripts are now completely automated for roughly 75-85% of all devices we service, they can spend more time doing hardware diagnosis & repair things instead of tediously typing command after command into a terminal, cutting and pasting results, etc.

We sincerely appreciate the work you've done that has allowed us to put this together.  We have more work to do, but the serialport stuff is pretty static at this point.  We have some changes we want to make to our wake instrument scripts, but nothing pressing as those changes are for a small percentage of our devices that connect via RS-485 instead of the norm of RS-232.

reconbot, 2017-06-21:
I'm dropping the precompiled builds for node4 win32 for any new versions of serialport going forward. There's no reason why it can't work if you build it however.