serialport/node-serialport #1087: Could not locate the bindings file

engirocha, 2017-02-09:
SerialPort version: 4.0.7
NodeJS Version: 6.5.0
Operating System and Hardware Platform: Windows 7 x86
Are you using an alternative NodeJS runtime? Electron

### Summary of Problem
Hi everyone, I can not run the electron without the error and list ports and devices connected.

> Uncaught Error: Could not locate the bindings file. Tried:
 → C:\Users\Demo\Desktop\app\node_modules\serialport\build\serialport.node
 → C:\Users\Demo\Desktop\app\node_modules\serialport\build\Debug\serialport.node
 → C:\Users\Demo\Desktop\app\node_modules\serialport\build\Release\serialport.node
 → C:\Users\Demo\Desktop\app\node_modules\serialport\out\Debug\serialport.node
 → C:\Users\Demo\Desktop\app\node_modules\serialport\Debug\serialport.node
 → C:\Users\Demo\Desktop\app\node_modules\serialport\out\Release\serialport.node
 → C:\Users\Demo\Desktop\app\node_modules\serialport\Release\serialport.node
 → C:\Users\Demo\Desktop\app\node_modules\serialport\build\default\serialport.node
 → C:\Users\Demo\Desktop\app\node_modules\serialport\compiled\6.5.0\win32\ia32\serialport.node

### Steps and Code to Reproduce the Issue

- Node.js 6.5.0
- NPM 3.10.10
- Python 2.7.13
- Windows build tools
- Node gyp

Here's our package.json:

`"dependencies": {
    "bindings": "^1.2.1",
    "data-table": "^2.0.0",
    "escpos": "^2.3.0",
    "serialport": "^4.0.7"
  }`

reconbot, 2017-02-09:
Please take a look at the readme for information about using Electron

engirocha, 2017-02-10:
Thank you very much, I installed the `electron-rebuild` and run `./node_modules/.bin/electron-rebuild.cmd` and `./node_modules/.bin/electron .`

closed 2017-02-10 by engirocha