serialport/node-serialport #1090: Issues running in Docker on Raspberry Pi

ArunKalahasti, 2017-02-11:
SerialPort version: 4.0.7
NodeJS Version: 4.2.1
Operating System and Hardware Platform: armhf flavor of alpine linux 3.5 running in docker 1.13.0 on a Raspberry Pi 3 Jessie

Are you using an alternative NodeJS runtime? (eg Electron):
node 7.5.0 compiled on the above platform

### Summary of Problem

### Steps and Code to Reproduce the Issue
listPorts.js
```javascript
var serialport = require('serialport');
 
// list serial ports:
serialport.list(function (err, ports) {
  ports.forEach(function(port) {
    console.log(JSON.stringify(port,undefined,4));
  });
});
```
package.json
```json
{
  "name": "serial-test",
  "version": "0.0.0",
  "description": "A test bed for the 'serialport' library",
  "main": "index.js",
  "scripts": {
    "test": "echo \"Error: no test specified\" && exit 1"
  },
  "author": "Arun Kalahasti",
  "license": "ISC",
  "dependencies": {
    "serialport": "^4.0.7"
  }
}

```

#### terminal commands:
##### To install depenedcies and build serialport from source in a removable container
```bash
$ docker run --device /dev/ttyACM0:/dev/ttyACM0 -it --rm -v "$PWD":/usr/src/app/ -w /usr/src/app/ arunk3001/armhf-node:7.5.0-alpine sh -c 'apk add --no-cache make gcc g++ python linux-headers udev && npm install serialport --build-from-souce=serialport'
```

##### returns no output
```bash
$ docker run --device /dev/:/dev/ -it --rm -v "$PWD":/usr/src/app/ -w /usr/src/app/ arunk3001/armhf-node:7.5.0-alpine node listPorts.js
```

##### Running without docker
```bash
$ node listPorts.js
{
    "comName": "/dev/ttyACM0",
    "manufacturer": "Arduino__www.arduino.cc_",
    "serialNumber": "Arduino__www.arduino.cc__0043_7543932373535141E0B2",
    "pnpId": "usb-Arduino__www.arduino.cc__0043_7543932373535141E0B2-if00",
    "vendorId": "0x2341",
    "productId": "0x0043"
}
{
    "comName": "/dev/ttyAMA0"
}
```

##### most progress I made
```bash
$ docker run --device /dev/ttyACM0:/dev/ttyACM0 -it --rm -v "$PWD":/usr/src/app/ -w /usr/src/app/ arunk3001/armhf-node:7.5.0-alpine node listPorts.js
(node:1) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 1): TypeError: Cannot read property 'forEach' of undefined
(node:1) DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

```

ArunKalahasti, 2017-02-11:
I've made some further progress.

I realized that some of the alpine dependencies might be required at run time, not just for building so I made an image which kept those dependencies as well as the npm install.
```bash
$ docker run -w /usr/src/app/ arunk3001/armhf-node:7.5.0-alpine sh -c 'apk add --no-cache make gcc g++ python linux-headers udev && npm install serialport --build-from-souce=serialport'
$ docker commit armhf-node-serialport
```

Now I can do:
```bash
$ docker run -v /dev:/dev -it --rm -v "$PWD":/usr/src/app/ -w /usr/src/app/ armhf-node-serialport sh -c 'node listPorts.js'
# output:
{
    "comName": "/dev/ttyACM0"
}
{
    "comName": "/dev/ttyAMA0"
}
```

Unfortunately, this output is still missing the details run without docker.

reconbot, 2017-02-11:
We use the commonly available program `udevadm` to query details on each port.

ArunKalahasti, 2017-02-11:
It seems some application in the stack is not able to show the information using udevadm. I'm not sure if that's docker, alpine, or some configuration I'm getting wrong. Either way, it seems I have gotten working results using the docker image I built and the aforementioned list, but connecting to and reading from a serial port required me to run the container with the `--privileged` flag on docker. Using the `--device` flag with a specific com port works, but I had been hoping to discover the expected connection as part of my startup process.

I'm leaving a link to the docker image in case it proves useful to someone hunting for help later:
https://hub.docker.com/r/arunk3001/armhf-node-serialport/

closed 2017-02-11 by ArunKalahasti

ArunKalahasti, 2017-02-11:
As you guys surely have more experience with serial connections than me, can I ask if there is a way to manually fetch device information based on comName?

reconbot, 2017-02-13:
I use `udevadm` to do that `¯\_(ツ)_/¯`

siburny, 2017-06-29:
I have the same issues. Here is the output of the udevadm command that this library runs `udevadm info --query=property -p $(udevadm info -q path -n /dev/ttyUSB0)`:
```
DEVNAME=/dev/ttyUSB0
DEVPATH=/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5/1-1.5:1.0/ttyUSB0/tty/ttyUSB0
MAJOR=188
MINOR=0
SUBSYSTEM=tty
```
There are no extra attributes returned. After some digging I found the workaround: start your container with `-v /run/udev:/run/udev:ro` - it gives read-only access to udev data. Now all works as expected:
```
DEVLINKS=/dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A6028MS6-if00-port0 /dev/serial/by-path/platform-3f980000.usb-usb-0:1.5:1.0-port0
DEVNAME=/dev/ttyUSB0
DEVPATH=/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5/1-1.5:1.0/ttyUSB0/tty/ttyUSB0
ID_BUS=usb
ID_MODEL=FT232R_USB_UART
ID_MODEL_ENC=FT232R\x20USB\x20UART
ID_MODEL_FROM_DATABASE=FT232 USB-Serial (UART) IC
ID_MODEL_ID=6001
ID_PATH=platform-3f980000.usb-usb-0:1.5:1.0
ID_PATH_TAG=platform-3f980000_usb-usb-0_1_5_1_0
ID_REVISION=0600
ID_SERIAL=FTDI_FT232R_USB_UART_A6028MS6
ID_SERIAL_SHORT=A6028MS6
ID_TYPE=generic
ID_USB_DRIVER=ftdi_sio
ID_USB_INTERFACES=:ffffff:
ID_USB_INTERFACE_NUM=00
ID_VENDOR=FTDI
ID_VENDOR_ENC=FTDI
ID_VENDOR_FROM_DATABASE=Future Technology Devices International, Ltd
ID_VENDOR_ID=0403
MAJOR=188
MINOR=0
SUBSYSTEM=tty
TAGS=:systemd:
USEC_INITIALIZED=95939112
```