serialport/node-serialport #1091: A dynamic link library （DLL） initialization routine failed

govenlive, 2017-02-12:
SerialPort version:latest
NodeJS Version:7.0.0
Operating System and Hardware Platform:win10
Are you using an alternative NodeJS runtime? (eg Electron):electron

### Summary of Problem
When I run the electron, throw such as error!
### Steps and Code to Reproduce the Issue

the code:
`var SerialPort = require("serialport").SerialPort;
    var portName = 'COM1';
    var serialPort = new SerialPort(portName, {
        baudRate: 9600,
        dataBits: 8,
        parityL:'none',
        stopBits: 1,
        flowControl: false
    }, false);
    serialPort.list(function (err, ports) {
        ports.forEach(function(port) {
            console.log(port.comName);
            console.log(port.pnpId);
            console.log(port.manufacturer);
        });
    });`

reconbot, 2017-02-13:
Have you tried to rebuild for electron? We have an installation section of our readme which gives a few tips https://github.com/EmergingTechnologyAdvisors/node-serialport#electron

govenlive, 2017-02-13:
Yes,I try to run after packaging，it's normal, but the debug run under electron or DLL error,I has update the electron to 1.6.0,but the debug  run the node is normal

zilohumberto, 2017-02-18:
i have the same problem!

facetime88, 2017-06-02:
same problem here

hanfengcan, 2017-06-02:
look at here http://www.jianshu.com/p/696fbdeb5b8a

reconbot, 2017-06-20:
Did it work? I can't read the link to know what it says.

reconbot, 2017-07-05:
Taking a better look at the example code; `.list` is a method of the constructor not of the instance of a serialport. 

I'm going to close this issue due to it's age, but if you'd like to continue with it feel free to comment and we'll reopen.

closed 2017-07-05 by reconbot

datamweb, 2017-08-20:
Hi,
I have the same error!!

reconbot, 2017-08-20:
Hi @datamweb please open a new issue with details on your setup

Quzal, 2017-09-29:
same error

Caumaker, 2017-11-19:
same erro here

hanfengcan, 2017-11-20:
@Caumaker @Quzal @datamweb 

a easy way to do

```
export npm_config_target=1.2.3
export npm_config_arch=x64
export npm_config_target_arch=x64
export npm_config_disturl=https://atom.io/download/electron
export npm_config_runtime=electron
export npm_config_build_from_source=true
```

if Windows  , export  use set, like
set npm_config_target=1.2.3

target is electron version

and then

`npm install serialport --save`

good luck

datamweb, 2017-12-08:
Hi,@hanfengcan help me step to step.please

hanfengcan, 2017-12-08:
@datamweb 
1. `export` that env_variables if you use linux. or use `set`  in windows.
2. use `npm install serialport --save`

datamweb, 2017-12-08:
@hanfengcan Thanks for your reply.
But ...
[image]

hanfengcan, 2017-12-08:
make sure you can build in nodejs. 
rebuild in electron

build a new electron project, if you use vue, you can create new project by https://github.com/SimulatedGREG/electron-vue 
And add serialport code to main/index.js