serialport/node-serialport #1094: I can't download serialport on my raspberry pi 3

sergioibs, 2017-02-14:
- __SerialPort Version__: serialport@4.0.7
- __NodeJS Version__: 1.4.21
- __Operating System__ and __Hardware Platform__:

> PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
> NAME="Raspbian GNU/Linux"
> VERSION_ID="8"
> VERSION="8 (jessie)"
> ID=raspbian
> ID_LIKE=debian

- Are you using __Electron__ and have you checked the __docs?__: No

### Summary of Problem

When I try to do this from my raspberry pi 3:
`sudo npm install serialport`

Next error appear:

> serialport@4.0.7 install /home/pi/Documents/serialLog/node_modules/serialport
> node-pre-gyp install --fallback-to-build

```
node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v11-linux-arm.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and node@0.10.29 (node-v11 ABI) (falling back to source compile with node-gyp) 
make: Entering directory '/home/pi/Documents/serialLog/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
In file included from ../src/./serialport.h:7:0,
                 from ../src/serialport.cpp:1:
../node_modules/nan/nan.h:324:47: error: ‘REPLACE_INVALID_UTF8’ is not a member of ‘v8::String’
   static const unsigned kReplaceInvalidUtf8 = v8::String::REPLACE_INVALID_UTF8;
                                               ^
serialport.target.mk:84: recipe for target 'Release/obj.target/serialport/src/serialport.o' failed
make: *** [Release/obj.target/serialport/src/serialport.o] Error 1
make: Leaving directory '/home/pi/Documents/serialLog/node_modules/serialport/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/share/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:809:12)
gyp ERR! System Linux 4.4.34-v7+
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "build" "--fallback-to-build" "--module=/home/pi/Documents/serialLog/node_modules/serialport/build/Release/serialport.node" "--module_name=serialport" "--module_path=/home/pi/Documents/serialLog/node_modules/serialport/build/Release"
gyp ERR! cwd /home/pi/Documents/serialLog/node_modules/serialport
gyp ERR! node -v v0.10.29
gyp ERR! node-gyp -v v0.12.2
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute 'node-gyp build --fallback-to-build --module=/home/pi/Documents/serialLog/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/home/pi/Documents/serialLog/node_modules/serialport/build/Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/pi/Documents/serialLog/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
node-pre-gyp ERR! stack     at maybeClose (child_process.js:755:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:822:5)
node-pre-gyp ERR! System Linux 4.4.34-v7+
node-pre-gyp ERR! command "node" "/home/pi/Documents/serialLog/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/pi/Documents/serialLog/node_modules/serialport
node-pre-gyp ERR! node -v v0.10.29
node-pre-gyp ERR! node-pre-gyp -v v0.6.32
node-pre-gyp ERR! not ok 
Failed to execute 'node-gyp build --fallback-to-build --module=/home/pi/Documents/serialLog/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/home/pi/Documents/serialLog/node_modules/serialport/build/Release' (1)
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm ERR! serialport@4.0.7 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@4.0.7 install script.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! System Linux 4.4.34-v7+
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install" "serialport"
npm ERR! cwd /home/pi/Documents/serialLog
npm ERR! node -v v0.10.29
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/Documents/serialLog/npm-debug.log
npm ERR! not ok code 0

```

reconbot, 2017-02-14:
I bet you have an old gcc. I also recommend using a newer version of node
js. 4x and 6x both run great on the rpi. The stock packages that ship with
Raspbian are very old

---
Francis Gulotta
wizard@roborooter.com

sergioibs, 2017-02-14:
Now
node version: 7.5
npm version: 4.1.2
gcc version: 4.9.2

Which could be the problem?

reconbot, 2017-02-14:
Now you're hitting the root problem https://github.com/EmergingTechnologyAdvisors/node-serialport#sudo--root

sergioibs, 2017-02-14:
But when I perform:
`pi@raspberrypi:~/Documents/serialLog $ sudo npm install serialport --unsafe-perm --build-from-source`

This is the result:

```
> serialport@4.0.7 install /home/pi/Documents/serialLog/node_modules/serialport
> node-pre-gyp install --fallback-to-build

make: Entering directory '/home/pi/Documents/serialLog/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
../src/serialport.cpp: In function ‘int getIntFromObject(v8::Local<v8::Object>, std::string)’:
../src/serialport.cpp:90:52: warning: ‘v8::Local<v8::Int32> v8::Value::ToInt32() const’ is deprecated (declared at /root/.node-gyp/7.5.0/include/node/v8.h:8442): Use maybe version [-Wdeprecated-declarations]
   return getValueFromObject(options, key)->ToInt32()->Int32Value();
                                                    ^
../src/serialport.cpp: In function ‘double getDoubleFromObject(v8::Local<v8::Object>, std::string)’:
../src/serialport.cpp:102:53: warning: ‘v8::Local<v8::Number> v8::Value::ToNumber() const’ is deprecated (declared at /root/.node-gyp/7.5.0/include/node/v8.h:8406): Use maybe version [-Wdeprecated-declarations]
   return getValueFromObject(options, key)->ToNumber()->NumberValue();
                                                     ^
../src/serialport.cpp: In function ‘void EIO_AfterOpen(uv_work_t*)’:
../src/serialport.cpp:174:31: warning: ‘v8::Local<v8::Int32> v8::Value::ToInt32() const’ is deprecated (declared at /root/.node-gyp/7.5.0/include/node/v8.h:8442): Use maybe version [-Wdeprecated-declarations]
     int fd = argv[1]->ToInt32()->Int32Value();
                               ^
../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Update(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/serialport.cpp:193:29: warning: ‘v8::Local<v8::Int32> v8::Value::ToInt32() const’ is deprecated (declared at /root/.node-gyp/7.5.0/include/node/v8.h:8442): Use maybe version [-Wdeprecated-declarations]
   int fd = info[0]->ToInt32()->Int32Value();
                             ^
../src/serialport.cpp:217:116: warning: ‘v8::Local<v8::Int32> v8::Value::ToInt32() const’ is deprecated (declared at /root/.node-gyp/7.5.0/include/node/v8.h:8442): Use maybe version [-Wdeprecated-declarations]
   baton->baudRate = Nan::Get(options, Nan::New<v8::String>("baudRate").ToLocalChecked()).ToLocalChecked()->ToInt32()->Int32Value();
                                                                                                                    ^
../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Write(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/serialport.cpp:252:29: warning: ‘v8::Local<v8::Int32> v8::Value::ToInt32() const’ is deprecated (declared at /root/.node-gyp/7.5.0/include/node/v8.h:8442): Use maybe version [-Wdeprecated-declarations]
   int fd = info[0]->ToInt32()->Int32Value();
                             ^
../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Close(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/serialport.cpp:369:32: warning: ‘v8::Local<v8::Int32> v8::Value::ToInt32() const’ is deprecated (declared at /root/.node-gyp/7.5.0/include/node/v8.h:8442): Use maybe version [-Wdeprecated-declarations]
   baton->fd = info[0]->ToInt32()->Int32Value();
                                ^
../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Flush(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/serialport.cpp:480:29: warning: ‘v8::Local<v8::Int32> v8::Value::ToInt32() const’ is deprecated (declared at /root/.node-gyp/7.5.0/include/node/v8.h:8442): Use maybe version [-Wdeprecated-declarations]
   int fd = info[0]->ToInt32()->Int32Value();
                             ^
../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Set(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/serialport.cpp:528:29: warning: ‘v8::Local<v8::Int32> v8::Value::ToInt32() const’ is deprecated (declared at /root/.node-gyp/7.5.0/include/node/v8.h:8442): Use maybe version [-Wdeprecated-declarations]
   int fd = info[0]->ToInt32()->Int32Value();
                             ^
../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Drain(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/serialport.cpp:585:29: warning: ‘v8::Local<v8::Int32> v8::Value::ToInt32() const’ is deprecated (declared at /root/.node-gyp/7.5.0/include/node/v8.h:8442): Use maybe version [-Wdeprecated-declarations]
   int fd = info[0]->ToInt32()->Int32Value();
                             ^
  CXX(target) Release/obj.target/serialport/src/serialport_unix.o
../src/serialport_unix.cpp: In function ‘OpenBatonPlatformOptions* ParsePlatformOptions(const v8::Local<v8::Object>&)’:
../src/serialport_unix.cpp:44:109: warning: ‘v8::Local<v8::Int32> v8::Value::ToInt32() const’ is deprecated (declared at /root/.node-gyp/7.5.0/include/node/v8.h:8442): Use maybe version [-Wdeprecated-declarations]
   result->vmin = Nan::Get(options, Nan::New<v8::String>("vmin").ToLocalChecked()).ToLocalChecked()->ToInt32()->Int32Value();
                                                                                                             ^
../src/serialport_unix.cpp:45:111: warning: ‘v8::Local<v8::Int32> v8::Value::ToInt32() const’ is deprecated (declared at /root/.node-gyp/7.5.0/include/node/v8.h:8442): Use maybe version [-Wdeprecated-declarations]
   result->vtime = Nan::Get(options, Nan::New<v8::String>("vtime").ToLocalChecked()).ToLocalChecked()->ToInt32()->Int32Value();
[... cut ...]
../src/serialport_poller.cpp:86:31: warning: ‘v8::Local<v8::Int32> v8::Value::ToInt32() const’ is deprecated (declared at /root/.node-gyp/7.5.0/include/node/v8.h:8442): Use maybe version [-Wdeprecated-declarations]
   obj->fd_ = info[0]->ToInt32()->Int32Value();
                               ^
  SOLINK_MODULE(target) Release/obj.target/serialport.node
  COPY Release/serialport.node
make: Leaving directory '/home/pi/Documents/serialLog/node_modules/serialport/build'
serial-log@1.0.0 /home/pi/Documents/serialLog
└── serialport@4.0.7 

npm WARN serial-log@1.0.0 No description
npm WARN serial-log@1.0.0 No repository field.
pi@raspberrypi:~/Documents/serialLog $ 

```

reconbot, 2017-02-14:
Those warnings are ok, it worked!

---
Francis Gulotta
wizard@roborooter.com

sergioibs, 2017-02-14:
Ah ok!,
Thank you!

closed 2017-02-14 by sergioibs