serialport/node-serialport #1096: Windows docker container build fails when building serialport

Xandroid4Net, 2017-02-14:
- __SerialPort Version__:  4.0.7
- __NodeJS Version__: 3.2.0
- __Operating System__ and __Hardware Platform__: WINDOWS 10 x64
- Are you using __Electron__ and have you checked the __docs?__: Not using electron

### Summary of Problem

docker container build fails with the trace below. I dont need to build it myself, but the docker build process looks for the following and it doesn't find it.

node-pre-gyp info it worked if it ends with ok
node-pre-gyp info using node-pre-gyp@0.6.32
node-pre-gyp info using node@3.2.0 | win32 | x64
node-pre-gyp info check checked for "C:\app\node_modules\serialport\build\Release\serialport.node" (not found)
node-pre-gyp http GET https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v
4.0.7-node-v45-win32-x64.tar.gz
node-pre-gyp http 404 https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v
4.0.7-node-v45-win32-x64.tar.gz
node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/downloa
d/4.0.7/serialport-v4.0.7-node-v45-win32-x64.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and node@3.2.0 (node-v45 ABI) (falling back to sourc
e compile with node-gyp)
node-pre-gyp http 404 status code downloading tarball https://github.com/EmergingTechnologyAdvisors/node-serialport/rele
ases/download/4.0.7/serialport-v4.0.7-node-v45-win32-x64.tar.gz

------------Docker build fail log----------------------------------------

C:\app\node_modules\serialport>if not defined npm_config_node_gyp (node "C:\Program Files\iojs\node_modules\npm\bin\node
-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" build --fallback-to-build --module=C:\app\node_modules\serialport
\build\Release\serialport.node --module_name=serialport --module_path=C:\app\node_modules\serialport\build\Release )  el
se (node  build --fallback-to-build --module=C:\app\node_modules\serialport\build\Release\serialport.node --module_name=
serialport --module_path=C:\app\node_modules\serialport\build\Release )
gyp info it worked if it ends with ok
gyp info using node-gyp@2.0.2
gyp info using node@3.2.0 | win32 | x64
gyp info spawn C:\Program Files (x86)\MSBuild\14.0\bin\msbuild.exe
gyp info spawn args [ 'build/binding.sln',
gyp info spawn args   '/clp:Verbosity=minimal',
gyp info spawn args   '/nologo',
gyp info spawn args   '/p:Configuration=Release;Platform=x64' ]
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
  serialport.cpp
  serialport_win.cpp
..\src\serialport_win.cpp(178): warning C4311: 'type cast': pointer truncation from 'HANDLE' to 'int' [C:\app\node_modul
es\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(178): warning C4302: 'type cast': truncation from 'HANDLE' to 'int' [C:\app\node_modules\seria
lport\build\serialport.vcxproj]
..\src\serialport_win.cpp(200): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [C:\app\no
de_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(207): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [C:\app\no
de_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(217): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [C:\app\no
de_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(219): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [C:\app\no
de_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(223): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [C:\app\no
de_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(225): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [C:\app\no
de_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(229): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [C:\app\no
de_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(231): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [C:\app\no
de_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(236): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [C:\app\no
de_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(248): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [C:\app\no
de_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(363): warning C4311: 'type cast': pointer truncation from 'HANDLE' to 'int' [C:\app\node_modul
es\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(363): warning C4302: 'type cast': truncation from 'HANDLE' to 'int' [C:\app\node_modules\seria
lport\build\serialport.vcxproj]
..\src\serialport_win.cpp(373): warning C4311: 'type cast': pointer truncation from 'HANDLE' to 'int' [C:\app\node_modul
es\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(373): warning C4302: 'type cast': truncation from 'HANDLE' to 'int' [C:\app\node_modules\seria
lport\build\serialport.vcxproj]
..\src\serialport_win.cpp(385): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [C:\app\no
de_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(410): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [C:\app\no
de_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(424): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [C:\app\no
de_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(451): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [C:\app\no
de_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(453): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [C:\app\no
de_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(567): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [C:\app\no
de_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(576): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [C:\app\no
de_modules\serialport\build\serialport.vcxproj]
  win_delay_load_hook.c
C:\Program Files\iojs\node_modules\npm\node_modules\node-gyp\src\win_delay_load_hook.c(31): error C2373: '__pfnDliNotify
Hook2': redefinition; different type modifiers [C:\app\node_modules\serialport\build\serialport.vcxproj]
  C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\include\delayimp.h(134): note: see declaration of '__pfnDliNoti
fyHook2'
gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\MSBuild\14.0\bin\msbuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Program Files\iojs\node_modules\npm\node_modules\node-gyp\lib\build.js:269
:23)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Windows_NT 10.0.14393
gyp ERR! command "C:\\Program Files\\iojs\\node.exe" "C:\\Program Files\\iojs\\node_modules\\npm\\node_modules\\node-gyp
\\bin\\node-gyp.js" "build" "--fallback-to-build" "--module=C:\\app\\node_modules\\serialport\\build\\Release\\serialpor
t.node" "--module_name=serialport" "--module_path=C:\\app\\node_modules\\serialport\\build\\Release"
gyp ERR! cwd C:\app\node_modules\serialport
gyp ERR! node -v v3.2.0
gyp ERR! node-gyp -v v2.0.2
gyp ERR! not ok

### Steps and Code to Reproduce the Issue

There are quite a few steps to setup the docker build environment on windows so repro steps are hard to add.

reconbot, 2017-02-14:
We don't support node@3.2.0, I'd use 4 or 6

Xandroid4Net, 2017-02-15:
Ok, thanks.

closed 2017-02-15 by Xandroid4Net

ArunKalahasti, 2017-03-04:
Apologies, Xandroid4Net, but could you confirm if changing the version worked for you?