serialport/node-serialport #1097: Send a file to arduino

antmag, 2017-02-16:
hey everybody,

I would like to send a file to arduino via serial port (GCODE)
The connection is OK and I can send the file but after a certain number of lines the process stop.
It is because the size of the file is too long.
Is there an option or a parameter to change to send bigger files than that?

Thanks in advance

antmag, 2017-02-17:
port6.write("$x \r\n", function(err) {
    if (err) {
      return console.log('Error on write: ', err.message);
    }
    console.log('message written');
  });

  port6.on('data', (data) => {
   console.log(data);
 })
  
  fs.readFile("C:\\...\\PORTAL1_COM6.nc", 'utf8', function (err,data) {
    if (err) {
      return console.log(err);
    }

    port6.write(data, function(err) {
      if (err) {
        return console.log('Error on write: ', err.message);
      }
      console.log('message written');
    });

    port6.on('data', (data) => {
     console.log(data);
   })

  });

reconbot, 2017-03-02:
What error are you getting? What platform are you using?

antmag, 2017-03-09:
No error! But im sending a gcode (150 lines) from my dashboard to my arduino (GRBL)
It was working perfectly with universal gcode sender and now with serialport package it send 15 lines and stop the gcode reading (each time stop at the same line). I think it is because my buffer is full and he cannot stock 150 lines. Do you have any idea about this issue?

antmag, 2017-03-09:
I tryed with python too and that was the same thing (gcode stop after 15 lines) 
so im looking to send 6 lines by 6 lines but I dont find the way to do that

reconbot, 2017-07-05:
How you write the lines shouldn't matter. It should buffer and then send when it can. Possibly the arduino has stopped reading?

I'm going to close this issue due to it's age, but if you'd like to continue with it feel free to comment and we'll reopen.

closed 2017-07-05 by reconbot