serialport/node-serialport #1101: npm install Error: Unsupported target version: 0.18.5 on Win10

MartinBspheroid, 2017-02-18:
After two days of trying to install this on my own, I have to open this issue. 

`npm i serialport `

gives [me these errors 
](https://gist.github.com/MartinBspheroid/1903afc9bfae5683e290ce16d5f86532)

which boils down to this line:
`node-pre-gyp ERR! stack Error: Unsupported target version: 0.18.5` 

I tried different setup with node-pre-gyp, several times clean up and start again, still stuck at same point. Biggest problem for me is that I cannot google anything related to this error.
Thanks!
Martin

-->
- __SerialPort Version__: current (through npm)
- __NodeJS Version__: 6.9.5 (tried 7.5.0)
- __Operating System__ and __Hardware Platform__: Windows 10
- Are you using __Electron__ and have you checked the __docs?__: yep, back and forwards

### Summary of Problem

### Steps and Code to Reproduce the Issue

reconbot, 2017-02-18:
Can you give more of the output including the install command?

MartinBspheroid, 2017-02-18:
Ok, [I updated gist with error message, ](https://gist.github.com/MartinBspheroid/1903afc9bfae5683e290ce16d5f86532)now with complete logs from "npm i serialport --save" all the way to the end 

And here is [log file](https://gist.github.com/MartinBspheroid/519468641193b0c2de7d316f80290e99)

reconbot, 2017-02-18:
Are you actually using electron? I don't see any references in your logs. This is a new one for me too.

MartinBspheroid, 2017-02-18:
No, maybe later I'll try that as well, but currently I'm fine with generic
node.js usage. Although I have Electron installed in global scope, I think
there's no chance it could interfere with node instalation.

Strangely I have no idea what refers to version  0.18.5 exactly.  I tried
search in node-gyp and node-pre-gyp, as well as serialport on github,  but
there is not a single notice of anything like this.

reconbot, 2017-02-18:
Do a `node -p process.versions` for me?

MartinBspheroid, 2017-02-18:
I'm not in front of computer now, so first thing tomorrow.

Thanks for fast response
Martin

MartinBspheroid, 2017-02-19:
<pre><code>PS C:\Dev\_jsPrototypes\serialPort> node -p process.versions
{ http_parser: '2.7.0',
  node: '6.9.5',
  v8: '5.1.281.89',
  uv: '1.9.1',
  zlib: '1.2.8',
  ares: '1.10.1-DEV',
  icu: '57.1',
  modules: '48',
  openssl: '1.0.2k' }
PS C:\Dev\_jsPrototypes\serialPort>
</code></pre>

reconbot, 2017-02-19:
Well your error is from here https://github.com/mapbox/node-pre-gyp/blob/master/lib/util/versioning.js#L156

Which means the node abi target version is being reported as `0.18.5`? That doesn't make sense. I'm having a hard time tracing how it gets called. I think maybe you should open an issue with node-pre-gyp.

MartinBspheroid, 2017-02-20:
alright, I opened new issue [here](https://github.com/mapbox/node-pre-gyp/issues/266), so let's see

closed 2017-03-02 by reconbot