serialport/node-serialport #1103: %1 is not a valid Win32 application

govenlive, 2017-02-20:
When in doubt it never hurts to ask!

Finally if you're opening an issue fill out the form below!

Thanks!
-->
- __SerialPort Version__ 4.0.7:
- __NodeJS Version__ 7.5.0   but  I'm electron1.6.0:
- __Operating System__ WIN10 x64 

### Summary of Problem
[image]

the error  

[image]

reconbot, 2017-02-20:
What are you doing to get the error?

govenlive, 2017-02-21:
I need to compile for 32-bit programs running, but my computer is 64 bit WIN10, such as the build - win - ia32, then an error, if compiling for a 64 - bit (build - win -- x64) is normal

reconbot, 2017-02-21:
You can download 32 bit builds from our releases page. https://github.com/EmergingTechnologyAdvisors/node-serialport/releases

closed 2017-06-20 by reconbot