serialport/node-serialport #1104: Windows nodejs serialport don't opens port

karenYerevan, 2017-02-20:
Windows nodejs serialport don't opens port

var SerialPort = require("serialport");
var port = new SerialPort("COM5");

port.on('open', function() {
  port.write('main screen turn on', function(err) {
    if (err) {
      return console.log('Error on write: ', err.message);
    }
    console.log('message written');
  });
});

// open errors will be emitted as an error event 
port.on('error', function(err) {
  console.log('Error: ', err.message);
})

Opening COM5: Unknown error code 121

reconbot, 2017-02-25:
Can you give us a little bit more information?

- Operating System and Hardware: 
- NodeJS Version: 
- SerialPort version: 

Thanks!

mikew, 2017-05-04:
I'm seeing the same error, on Windows. Unable to open ports with `Unknown error code 121`.

Sadly I cannot reproduce locally and this is an issue with a client an ocean away.

[image]

reconbot, 2017-06-20:
We see this sometimes with some serialport drivers. Not sure what to do about it, upgrading the driver has helped.

closed 2017-06-20 by reconbot