serialport/node-serialport #1106: mac OSX Sierra   only have Bluetooth port, but, not the USB ports

heuuLZP, 2017-02-23:
I am using below code to list the available ports using node-serialport npm:

SerialPort.list(function (err, ports) {
    ports.forEach(function(port) {
      console.log(port);
    });
  });

It lists only the Bluetooth port, but, not the USB ports

Below is the output that I am getting:

[ { comName: '/dev/cu.Bluetooth-Incoming-Port',
    manufacturer: undefined,
    serialNumber: undefined,
    pnpId: undefined,
    locationId: undefined,
    vendorId: undefined,
    productId: undefined } ]

heuuLZP, 2017-02-25:
If you have the some question, you can refer to my practice.

I install the drive of usb, and will get it message.

{ comName: '/dev/cu.SLAB_USBtoUART',
  manufacturer: 'Silicon Labs',
  serialNumber: '0001',
  pnpId: undefined,
  locationId: '0x14200000',
  vendorId: '0x10c4',
  productId: '0xea60' }

closed 2017-02-25 by heuuLZP

reconbot, 2017-02-25:
Glad the usb virtual serialport drivers worked out for you.