serialport/node-serialport #1107: please confirm if "node-pre-gyp" can be removed when deploy

gengshenghong, 2017-02-25:
Hi, this is a general question about "node-pre-gyp" in serialport library, in my installation of serialport, there is a "node_modules" in it and folders as below:
bindings
node-pre-gyp

node-pre-gyp has a very long/deep path, which will cause some issues on windows (the max path is 255 usually and when I deploy serialport, the left path length for user is very little.

Please confirm whether it is safe that I can remove the "node-pre-gyp" directly when deploy. I've tried to remove it and from my test everything works well, but hope to get solid confirmation as I am not quite sure.

Thanks,
Shenghong

reconbot, 2017-02-26:
`npm3` or `npm4` will solve your path issues. If you already have the binary installed you can safely remove `node-pre-gyp`. It is only used during install time.

gengshenghong, 2017-02-28:
Thank you for your confirmation. I'll remove it manually as of now.

closed 2017-06-20 by reconbot