serialport/node-serialport #1111: How to close the port

jin-pan, 2017-03-02:
In my page,i use

```
var SerialPort = require('serialport');
var port = new SerialPort('COM2', function (err) {
  if (err) {
    console.log('Error: ', err.message);
  }
  else{
	console.log('Success.');
  }
});
```

to new a serialport,when i load this page first time,It console.log ("Success").

But If refresh this page,It console.log( "Error"). 
And I know the reason is 'COM2' is not close.It issue fail when it new SerialPort().

And my question is:

	how can i close the 'COM2' when the page is reload?
	or has there a way to renew this 'COM2' to var port?

	
Thank You.

reconbot, 2017-03-02:
`port.close()` will close the port. I'm assuming you're using electron? You might want to look at the[`close event`](https://electron.atom.io/docs/api/browser-window/#event-close). You'll want to use electron's github for support with electron.

jin-pan, 2017-03-03:
@reconbot  thank you your answer.

I know use `port.close()` to close the port.
and my problem is in my page I use
`var port = new SerialPort()`

When the page is load on first time, It can new a serialport to variable "port" success.
When we reload the page again, `var port = new SerialPort()` will be issue a Error,an at this case,when use port.close(); will issue an "close() function is not defined" error due to fail to New a serialport.

But I solve this case:

i use a global variable "port" to new the SerialPort, which variable "port" was not be initialized during the page reload.
In this case, when the page reload,the variable "port" still cantain the serialport Object.
So,port.close(),is done.

thank you again.

closed 2017-03-03 by jin-pan