serialport/node-serialport #1114: Windows: No event if a serial/bluetooth device is switched off.

soekdd, 2017-03-08:
- __SerialPort Version__: 4.0.7
- __NodeJS Version__: 6.9.4 
- __Operating System__ and __Hardware Platform__: Windows7/x64
- Have you checked the right __version__ of the __api docs?__: I checked all the docs

### Summary of Problem

The modules doesn't realize if a serial via bluetooth connection is switched off. Non off the "error", "disconnected" or "closed" event listener is reacting. 

I uses bluetooth devices which sometimes need to change the battery. My node,js program isn't realizing that the connection is off - so i cannot reconnect automatically.

### Steps and Code to Reproduce the Issue
- connect any serial via bluetooth device on windows, like some barcode readers or other devices
- connect via nodes/node-serialport
- establish event listener on "error", "disconnected" and "closed"
- switch off the device physical
=> no event is thrown

Any change to fix it?

reconbot, 2017-03-11:
Are you trying to write to the port at all while it disconnects? We don't get OS level events for disconnection but we do get a list of errors on read or write.

soekdd, 2017-03-12:
Ok, so the solution would be to try a kind of a periodic test writes? I will try it out. Thanks for the reply!

closed 2017-03-12 by soekdd