serialport/node-serialport #1115: UPGRADE_GUIDE.md needs an update

willcook4, 2017-03-10:
- Any Version:
- n/a :
- n/a:

### Summary of Problem and Steps and Code to Reproduce the Issue
The file UPRADE_GUIDE.md needs an update. As someone who wants to get version 5, there is no info on how to go about this. The npm repositories only have the v4 releases.

Also, the issue 1046 is not linked correctly. Even though that appears to be a closed issue with no detail.

If it helps here is the section from UPGRADE_GUIDE.md with the issue url correctly linked.

```
Upgrading from 4.x to 5.x
-------------
5.x is a major rewrite to make node serialport a NodeJS stream. While the api surface is  similar there have been a number of changes to ensure more consistent error handling and operation of a serial port.

The exact changes will go [here](https://github.com/EmergingTechnologyAdvisors/node-serialport/issues/1046)

```

reconbot, 2017-03-11:
Happy to take a pr to fix the issue. 5x can be found under the beta tag. `npm install serialport@beta` or `npm install serialport@5`

reconbot, 2017-03-11:
And thank you for testing 5x!

willcook4, 2017-03-13:
Thanks I'll do some testing and see how far I get. Thanks for the update on where to find v5

reconbot, 2017-07-10:
dupe of #1046 still need to finish writing this now that we're gearing up for release.

closed 2017-07-10 by reconbot