serialport/node-serialport #1116: not works in electron

alexey2baranov, 2017-03-11:
hello! 

I get this error with version mismatch

> 
> A JavaScript error occurred in the main process
> Uncaught Exception:
> Error: The module '/home/alexey2baranov/htdocs/termilocal-js/node_modules/serialport/build/Release/serialport.node'
> was compiled against a different Node.js version using
> NODE_MODULE_VERSION 51. This version of Node.js requires
> NODE_MODULE_VERSION 53. Please try re-compiling or re-installing
> the module (for instance, using `npm rebuild` or`npm install`).
>     at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:173:20)
>     at Object.Module._extensions..node (module.js:598:18)
>     at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:173:20)
>     at Module.load (module.js:488:32)
>     at tryModuleLoad (module.js:447:12)
>     at Function.Module._load (module.js:439:3)
>     at Module.require (module.js:498:17)
>     at require (internal/module.js:20:19)
>     at bindings (/home/alexey2baranov/htdocs/termilocal-js/node_modules/bindings/bindings.js:76:44)
>     at Object.<anonymous> (/home/alexey2baranov/htdocs/termilocal-js/node_modules/serialport/lib/bindings.js:3:35)

when i execute this code from electron browser page
```js
console.log(JSON.stringify(process.versions)))
```
it returns
> {"http_parser":"2.7.0","node":"7.4.0","v8":"5.6.326.50","uv":"1.10.1","zlib":"1.2.8","ares":"1.10.1-DEV","modules":"53","openssl":"1.0.2j","electron":"1.6.2","chrome":"56.0.2924.87","atom-shell":"1.6.2"}

so i change my node version down to 7.4.0 to be equal with electron internal node version with 
`nvm install v7.4.0` now `node -v `
returns 
> v7.4.0

I removed all node_modules folder and execute `npm install` again. Unfortunately i get the same error as before. What else should i do ??

reconbot, 2017-03-11:
Please look at the readme about using electron rebuild

alexey2baranov, 2017-03-12:
electron-rebuild rulezzzz

closed 2017-03-12 by alexey2baranov