serialport/node-serialport #1119: SerialPort.list still get null

keyiis, 2017-03-19:
- __SerialPort Version__:4.0.7
- __NodeJS Version__:electron 1.4.15 ia32
- __Operating System__ and __Hardware Platform__:win10 x64

- Have you checked the right __version__ of the __api docs?__:yes！
- Are you having trouble installing and you checked the __Installation Special Cases__ docs?yes！
- Are you using __Electron__ and have you checked the _Electron Docs_?:yes！

### Summary of Problem
I used SerialPort with electron,but SerialPort.list still get null array.
### Steps and Code to Reproduce the Issue

```
SerialPort.list( (err: string, ports: SerialPort.portConfig[]) => {
            if(err) return reject(err);
            //still get null array
            resolve(ports);
        });
```

reconbot, 2017-03-19:
What's a null array? No array or an array with nulls? How many Nulls? one for each port?

In any case we've seen similar issues when the binary isn't built correctly with electron.

keyiis, 2017-03-20:
@reconbot 'null array' == [],today when i insert a serialport device into usb port,SerialPort.list can get a port [object],so SerialPort.list only get working port?

reconbot, 2017-03-20:
Yes, that's what it does. Lists the current ports on the system.

---
Francis Gulotta
wizard@roborooter.com

keyiis, 2017-03-21:
@reconbot thks reply! i have 2 question:

1.  if i want listen all SerialPort plug state on my computer,should i do as below

```
setTimeout(()=>{
     SerialPort.list( (err: string, ports: SerialPort.portConfig[]) => {
            if(err) return reject(err);
            console.log(ports);
        });
 },1000
);
```

2. if i second time wirte some to port,still get 'Error: Opening COM5: Access denied',below is code

```
wirteToPort('12345678');//success!
wirteToPort('87654321');//failed
function wirteToPort(msg:string){
     let port = new SerialPort('COM5', {
            baudRate:baudRate,
            autoOpen:false
        });
        if (port.isOpen()) {
            port.write(msg, werr => {
                if (werr) console.log(werr);
            });
        } else {
            port.open(err => {
                if (err)  console.log(err);
                port.write(msg, werr => {
                    if (werr)  console.log(werr);
                });
            });
        }
}
```

reconbot, 2017-03-21:
>    1. if i second time wirte some to port,still get 'Error: Opening COM5:

The polling is probably ok. But you should keep using the same port object.

keyiis, 2017-03-21:
@reconbot if myapp occur error after port.open(),then i havn't  unplugging and re-plugging the serialport after restart my app?

reconbot, 2017-03-21:
You might be able to close the port and then reopen it. Your example code
never closed the port and makes a new object for the same port. That's
probably your error.

keyiis, 2017-03-21:
@reconbot i update my code,second time wirte is ok,but if error occur before port.close() and i still need replug serialport.so has some way get a opened port through 'new SerialPort'?
```
public send(portPath: string, baudRate: number, cmd: string) {
        return new Promise((resolve, reject) => {
            let port = new SerialPort(portPath, {
                baudRate: baudRate,
                autoOpen: false
            });
            port.open(err => {
                if (err) return reject('open port' + err);
                port.write(cmd, werr => {
                    if (werr) return reject('wirte port' + werr);
                    port.close(cerr => {
                        if (cerr) return reject('close port' + cerr);
                        resolve();
                    });
                });
            });
        });
    }
```

reconbot, 2017-07-05:
This works on `serialport@5.0.0-beta6` let me know if you have any issues.

I'm going to close this issue due to it's age, but if you'd like to continue with it feel free to comment and we'll reopen.

closed 2017-07-05 by reconbot