serialport/node-serialport #1121: Uncaught TypeError: undefined is not a function

keyiis, 2017-03-23:
- __SerialPort Version__:4.0.7
- __NodeJS Version__:6.9.2
- __Operating System__ and __Hardware Platform__:win10 x64 electron 1.4.15 ia32

- Have you checked the right __version__ of the __api docs?__:yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs?nO
- Are you using __Electron__ and have you checked the _Electron Docs_?:yes

### Summary of Problem
When i unplug serialport device when my app runing ,console occur belows error in cycle:
`Uncaught TypeError: undefined is not a function`
[image]

i can't debug it,because it not stop.

keyiis, 2017-03-23:
I think maybe serialport unlimited try connect the port after i unplug device,i try to on('close')  and on('disconnect') but not tigger.

reconbot, 2017-03-23:
Windows might keep reading with errors. Should be fixed in 5x beta

keyiis, 2017-03-23:
@reconbot thks reply! i update to 5.0.0-beta3,it fixed!

closed 2017-03-23 by keyiis