serialport/node-serialport #1122: Error with electron/webpack/typescript/angular2 AOT

keyiis, 2017-03-24:
- __SerialPort Version__:5.0.3
- __NodeJS Version__:6.9.2
- __Operating System__ and __Hardware Platform__:win10 x64

- Have you checked the right __version__ of the __api docs?__:yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs?yes
- Are you using __Electron__ and have you checked the _Electron Docs_?:yes

### Summary of Problem
A strange question,serialport with webpack/typescript/angular2 work fine on development env,but not work on production env,development and  production has little different:
1、production use @ngtool-webpack compiler ts file and angular2 AOT.
2、production use webpack UglifyJsPlugin to minify js file. 
error:
```
Uncaught Error: Could not locate the bindings file. Tried:
→ C:\myapp\serialport\build\serialport.node
→ C:\myapp\serialport\build\Debug\serialport.node
→ C:\myapp\serialport\build\Release\serialport.node
→ C:\myapp\serialport\out\Debug\serialport.node
→ C:\myapp\serialport\Debug\serialport.node
→ C:\myapp\serialport\out\Release\serialport.node
→ C:\myapp\serialport\Release\serialport.node
→ C:\myapp\serialport\build\default\serialport.node
```
because my app can runing on devlopemnt env,so i think electron-rebuild work correctly,the reson maybe about production env webpack config.someone has exprerience about this?
**production env webpack config file**
```
var webpack = require('webpack');
const HtmlWebpackPlugin = require('html-webpack-plugin');
var webpackMerge = require('webpack-merge');
var ExtractTextPlugin = require('extract-text-webpack-plugin');
var commonConfig = require('./webpack.common.js');
var helpers = require('./helpers');
const AotPlugin = require('@ngtools/webpack').AotPlugin;

const ENV = process.env.NODE_ENV = process.env.ENV = 'production';

module.exports = webpackMerge(commonConfig, {
  entry: {
    'app': helpers.rootAngular('app/main.ts')
  },
  output: {
    path: helpers.outputBuild(),
    filename: '[name].[hash].bundle.js',
    chunkFilename: '[id].[chunkhash].chunk.js'
  },

  module: {
    loaders: [
      { test: /\.ts$/, loaders: ['@ngtools/webpack'] },
      {
        test: /\.css$/,
        exclude: helpers.rootAngular('app'),
        loader: ExtractTextPlugin.extract({ fallbackLoader: 'style-loader', loader: 'css-loader' })
      },
    ]
  },
  plugins: [
    // new OptimizeJsPlugin({
    //   sourceMap: false
    // }),
    new AotPlugin({
        tsConfigPath: helpers.root('tsconfig.aot.json'),
        entryModule: helpers.rootAngular('app/app.module#AppModule')
    }),
    new webpack.NoEmitOnErrorsPlugin(),
    new webpack.optimize.UglifyJsPlugin({
            beautify: false,
            mangle: {
                screw_ie8: true,
                keep_fnames: true
            },
            compress: {
                warnings: false,
                screw_ie8: true
            },
            comments: false
    }),
    new HtmlWebpackPlugin({
      template: helpers.rootAngular('index.html'),
      minify:{
        removeComments:true,
        collapseWhitespace: true
      }
    }),
    new ExtractTextPlugin('[name].[hash].css'),
    new webpack.DefinePlugin({
      'process.env': {
        'ENV': JSON.stringify(ENV)
      }
    })
  ]
});

```

reconbot, 2017-03-24:
The binary file, the .node cannot be webpacked (I haven't heard anyone
successfully​doing it due to the nature of binary module loading.) In Dev
you probably have your node modules folder around, in prod you probably
don't.

keyiis, 2017-03-24:
@reconbot i already set serialport as externals in webpack,i had other binary module used in project,they all work fine.
```
externals: {
    "electron":"require('electron')",
    "sequelize": "require('sequelize')",
    "moment":"require('moment')",
    "pinyin":"require('pinyin')",
    "sqlite3":"require('sqlite3')",
    "printer": "require('printer')",
    "serialport": "require('serialport')",
    "fs":"require('fs')"
  },
```

keyiis, 2017-03-26:
@reconbot Is serialport used object.assign?why i get 'could not find module object.assign',if i removed serialport package,the error disappeared

reconbot, 2017-03-26:
This exists in node 4+ only.

keyiis, 2017-03-28:
@reconbot  found the source of the issuse,if i removed 'webpack.optimize.UglifyJsPlugin' from webpack,the error(Uncaught Error: Could not locate the bindings file) disappeared,so UglifyJsPlugin do something is causing the error,UglifyJsPlugin is a minify code plugin of webpack,But I haven't found a solution.

keyiis, 2017-03-28:
@reconbot I guess it may be relationships with the issuse,  in win32.js `const binding = require('bindings')('serialport.node');`

keyiis, 2017-03-28:
**step1** serialport win32.js call `require('bindings')`,
**step2** bindings.js exports.getFileName use Error.prepareStackTrace get called filename
**step3** bindings hope get `node_modules\serialport\lib\bindings\win32.js`,but relly get `polyfills.dfedb3b31c2eb36392b9.bundle.js`.so get error module_root path.
**step4** why ` polyfills.dfedb3b31c2eb36392b9.bundle.js` call bindings.js?

` polyfills.dfedb3b31c2eb36392b9.bundle.js`  process by UglifyJsPlugin, is come from 
**polyfills.ts**
```
import 'core-js/es6';
import 'core-js/es7';
import 'zone.js/dist/zone';
import 'zone.js/dist/long-stack-trace-zone';
```

keyiis, 2017-03-28:
I think bindings.getFileName has issuse.
**node_modules\bindings\bindings.js** line 110
```
Error.prepareStackTrace = function (e, st) {
    for (var i = 0, l = st.length; i < l; i++) {
      fileName = st[i].getFileName();
      console.log('__filename', __filename);
      console.log('fileName', fileName);
      /*if (fileName !== __filename) {
        if (calling_file) {
            if (fileName !== calling_file) {
              return
            }
        } else {
          return
        }
      }*/
    }
  }
```
**console**
```
__filename D:\approot\output\build\node_modules\bindings\bindings.js
fileName file:///D:/approot/output/build/polyfills.f4ea2e16b3f7f56b98ef.bundle.js

__filename D:\approot\output\build\node_modules\bindings\bindings.js
fileName D:\approot\output\build\node_modules\bindings\bindings.js

__filename D:\work\nodejs\keypos\pos\output\build\node_modules\bindings\bindings.js
fileName D:\work\nodejs\keypos\pos\output\build\node_modules\serialport\lib\bindings\win32.js
```

keyiis, 2017-03-29:
@reconbot I think bindings getFileName method logic is flawed,Error.prepareStackTrace can redefine in other js file.now i used temporary solution to hacked win32.js,this only worked for windows32 system
```
const binding_path = require("path").join(__dirname, '../../build/Release/serialport.node');
const binding = require(binding_path);
```
i hope serialport can fix it!

reconbot, 2017-03-29:
This seems like an issue with your uglify rules. Can you exclude the
bindings modules or something?

---
Francis Gulotta
wizard@roborooter.com

closed 2017-06-20 by reconbot