serialport/node-serialport #1125: Intermittent corruption when creating new serialport() on already opened device

petertorelli, 2017-03-29:
- __4.0.7__:
- __7.4.0__:
- __macOS Sierra 10.12.3__ and __15" MBP 2013__:
- __Win10__ and __Some lame DELL Laptop__:

- Have you checked the right __version__ of the __api docs?__:
Yes.

- Are you having trouble installing and you checked the __Installation Special Cases__ docs?
No.

- Are you using __Electron__ and have you checked the _Electron Docs_?:
No.

### Summary of Problem
I intermittently see non-ASCII noise appearing on the inputs of devices after trying to create a new serialport object on that port to determine if it is openable.

### Steps and Code to Reproduce the Issue
I have many devices connected to my system (STM Nucleo, SiLabs Gecko, Arduino, RPI, PL2303). The devices all have serial monitors running in their firmware I wrote that read the incoming characters from the UART and respond to simple commands. They all respond with a unique text string to a text "query" command so that I can tell which device is which. Since they all share similar vendor and product IDs, I can't use those values to ID the device so I have to go through this send/ack process.

Here's my process:

1. serialport.list to get all of the ports
2. For each port, create a new serialport object with the comName (autoOpen is default true)
3. I use on('error') to skip devices that are locked (either because I already opened them previously or they are not one of my own devices).
4. on('open') I do the following: wait 2 sec -> write & drain `query\r\n` -> wait for reply or timeout -> check reply

This works swimmingly ~80% of the time. I see the text `query\r\n` going out, and I see `device-type-NNN` coming back, where NNN are my own unique codes (or I see a timeout if it isn't one of my devices, like the bluetooth device on my mac).

The other 20% of the time, I will receive an error message from my device. My firmware is programmed to return `invalid-command-?COMMAND` if it doesn't recognize the string it received. The 20% error responses look like `invalid-command-?<><><><><><><><>query` where the <> are non-printable hex bytes. This tells me the problem is that the input queue had some garbage sent to it before `query\r\n`, because it is in the middle of the error message (as opposed to the return string being corrupted).

I have narrowed it down to this: the only device that does NOT suffer this is the LAST device on the list. In other words, it seems like trying to create a new serialport object on a device that I have already previously opened is somehow corrupting the existing open connection.

Did that make sense? I sure hope so. It's maddening.

Thanks in advance,
P

reconbot, 2017-06-20:
Windows doesn't support locking the port. OSX does but I bet the `open` does some configuring commands that change the baud rate even if it can't open the port in the end.

reconbot, 2017-07-06:
I'm going to close this issue due to it's age, but if you'd like to continue with it feel free to comment and I'll reopen. I'd love to debug further.

closed 2017-07-06 by reconbot