serialport/node-serialport #1126: Connection to port established, but `port.write` and/or `data` event is not working

johnmanko, 2017-03-30:
- __SerialPort Version__: 4.0.7
- __NodeJS Version__: v7.4.0
- __Operating System__ and __Hardware Platform__: Ubuntu 16.10 and x86-64

- Have you checked the right __version__ of the __api docs?__: YES
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? NO
- Are you using __Electron__ and have you checked the _Electron Docs_?: NO

### Summary of Problem
Connection to port is successfully established, but `port.write` and/or `data` event is not working.
Added the current user to group `dailout` to correct problems with permissions.

### Steps and Code to Reproduce the Issue
```
var SerialPort = require('serialport');
var options = {
  autoOpen: false,
  lock: false,
  baudRate: 19200,
  dataBits: 8,
  stopBits: 1,
  parity: 'none',
  rtscts: false,
  xon: false,
  xoff: false,
  xany: false,
  bufferSize: 55
};

var comName = '/dev/ttyUSB0';

SerialPort.list(function (err, ports) {
  ports.forEach(function(p) {
    if(!p.vendorId) {return;}
    if(p.comName === comName) {
      console.log('**************** MATCHED PORT ****************');
      console.log(p);
      doStuff();
    }
  });
})

function doStuff() {

  var port = new SerialPort(comName, options);

  port.on('error', function(err) {
    console.log('Error: ', err.message);
  });  

  port.on('data', function (data) {
    console.log('Is still connected?', port.isOpen());
    console.log('Data: ' + data);
    port.close();
  });

  port.on('open', function () {
    console.log('**************** PORT OPENED *****************');
    console.log('Is connected?', port.isOpen());
    console.log('**************** PORT DETAILS ****************');
    console.log('OPEN PORT: ', port);
    var buffer = Buffer.from([0xFF, 0x01, 0x00, 0x02, 0xFE]);
    console.log('**************** BUFFER DETAILS **************');
    console.log('buffer.toString("hex") = ', buffer.toString("hex"));
    port.write(buffer);

  });
  port.open();
}
```
Here is the output:
```
**************** MATCHED PORT ****************
{ comName: '/dev/ttyUSB0',
  manufacturer: 'Prolific_Technology_Inc.',
  serialNumber: 'Prolific_Technology_Inc._USB-Serial_Controller_D_12345855',
  pnpId: 'usb-Prolific_Technology_Inc._USB-Serial_Controller_D_12345855-if00-port0',
  vendorId: '0x067b',
  productId: '0x2303' }
**************** PORT OPENED *****************
Is connected? true
**************** PORT DETAILS ****************
OPEN PORT:  SerialPort {
  domain: null,
  _events: { error: [Function], data: [Function], open: [Function] },
  _eventsCount: 3,
  _maxListeners: undefined,
  path: '/dev/ttyUSB0',
  fd: 12,
  paused: false,
  opening: false,
  closing: false,
  bufferSize: 55,
  readable: true,
  reading: false,
  options: 
   { baudRate: 19200,
     autoOpen: false,
     parity: 'none',
     xon: false,
     xoff: false,
     xany: false,
     rtscts: false,
     hupcl: true,
     dataBits: 8,
     stopBits: 1,
     bufferSize: 55,
     lock: true,
     parser: [Function: raw],
     platformOptions: { vmin: 1, vtime: 0 },
     disconnectedCallback: [Function: bound ],
     dataCallback: [Function: bound raw] },
  serialPoller: SerialportPoller {} }
**************** BUFFER DETAILS **************
buffer.toString("hex") =  ff010002fe
```

I don't have a manual, but I do have C# code (that works) I'm converting (from Windows-only to cross-platform) if that's helpful:

```
this.axMSComm.CommPort = 1; // COM1, COM2, COM3, etc
this.axMSComm.Settings = "19200,n,8,1";
this.axMSComm.PortOpen = true;
this.axMSComm.RThreshold = 55;
this.axMSComm.Output = new byte[] {255, 1, 0, 2, 254};
```

johnmanko, 2017-04-02:
Is **DTR** something that can **node-serialport** control?

[PL2303 Serial Port adapter doesn't work in either ubuntu 13:10 or 14:04](https://askubuntu.com/questions/465628/my-prolific-technology-inc-pl2303-serial-port-adapter-doesnt-work-in-either-u#469485):

> the usb-serial drivers for PL2303 do work and are not broken. I think most of the problems arise from the initialization where the DTR line defaults to -ve volts ie 0. Previous versions of Ubuntu (i.e. 10.04) the default was +ve volts on the DTR line. Note that other usb to serial converters such as Belkin F5U409 in UBUNTU 14.04 default to +ve volts on DTR line thus work OK. Confirmation of working drivers was done using GTKterm and toggling the DTR line. Furthermore connection to a Stanford Research SR830 Lock-in amplifier was established and its identify verified.
> 
> Having established working drivers, the solution in tcl/tk is to use the -ttycontrol switch within the fconfigure command. i.e.:
> ```
> ...
> set device "/dev/ttyUSB0"
> set dh [open $device r+]
> fconfigure $dh -mode $baud_rate,$parity,$data_bits,$stop_bits -translation binary -encoding binary -blocking 0 -buffering line -ttycontrol {RTS 0 DTR 1}
> set tmpstr [ read $dh ]
> close $dh 
> ...
> ```
>

johnmanko, 2017-04-02:
Well, it looks like `dtr` can be added as an option. Unfortunately, adding `dtr: true` to options doesn't resolve the problem. :(

https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/b80025ea1010d207610b1bc8cd97378f6d21bca8/src/serialport.cpp#L597

```
  SetBaton* baton = new SetBaton();
  memset(baton, 0, sizeof(SetBaton));
  baton->fd = fd;
  baton->callback.Reset(callback);
  baton->brk = getBoolFromObject(options, "brk");
  baton->rts = getBoolFromObject(options, "rts");
  baton->cts = getBoolFromObject(options, "cts");
  baton->dtr = getBoolFromObject(options, "dtr");
  baton->dsr = getBoolFromObject(options, "dsr");

  uv_work_t* req = new uv_work_t();
  req->data = baton;
uv_queue_work(uv_default_loop(), req, EIO_Set, (uv_after_work_cb)EIO_AfterSet);
```

https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/a049538270629299a17865d31dfc172338ebdd12/lib/bindings/base.js#L156
```
  /**
   * Set control flags on an open port.
   * @param {object=} options All options are operating system default when the port is opened. Every flag is set on each call to the provided or default values. All options will always be provided.
   * @param {Boolean} [options.brk=false]
   * @param {Boolean} [options.cts=false]
   * @param {Boolean} [options.dsr=false]
   * @param {Boolean} [options.dtr=true]
   * @param {Boolean} [options.rts=true]
   * @returns {Promise} Resolves once the port's flags have been set.
   * @throws {TypeError} When given invalid arguments a TypeError will be thrown.
   */
  set(options) {
    if (typeof options !== 'object') {
      throw new TypeError('"options" is not an object');
    }

    if (!this.isOpen) {
      return Promise.reject(new Error('Port is not open'));
    }
    return Promise.resolve();
}
```

reconbot, 2017-06-20:
Thank you for the detailed report, sorry it's been so long for a reply.

The code that would set DTR would be here https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/master/src/serialport_unix.cpp#L648-L688 (for linux and osx)

And the Javascript that calls it would be in the linux or osx bindings - you linked in the base bindings.

You do mention you're using serialport 4.0.7 and the code we've both linked to is in the 5x-beta series.

Were you able to figure this out?

johnmanko, 2017-07-08:
@reconbot I'll try again this weekend with the latest version and report back.

reconbot, 2017-07-08:
`serialport@5.0.0-beta6` is looking pretty good, it doesn't have any changes to `.set` or `.get` however.

johnmanko, 2017-07-23:
@reconbot no luck. Same result - no response. :(

reconbot, 2017-07-23:
I notice a few things in the example code that could cause errors. You call `port.open()` but the port is set to auto open, also you might have caught this but `port.isOpen` is a property in 5x instead of function.

I'm sorry I'm having trouble trying to follow the issue. So your javascript works on windows, but on linux the DTR control doesn't seem to work? Your original post says read and write doesn't work, but I think that's no longer the case?

I do recommend you test with 5x (soon to leave beta) and use the [debug output](https://www.npmjs.com/package/serialport#debugging) to ensure serialport is doing what you ask it to. eg `DEBUG=serialport:* node myapp.js`. 

As for `DTR` working on linux with that driver, I don't know.

reconbot, 2017-09-11:
I'm going to close this issue due to it's age, but if you'd like to continue with it feel free to comment and we'll reopen.

5x is out btw

closed 2017-09-11 by reconbot