serialport/node-serialport #1128: parser function called with only 1 to 10 bytes each time

farfromrefug, 2017-04-03:
With serialport 4.0.7 and 5.0.0.beta3 the parser function is called with very little arrays.

So i use serialport on both sides : Pi and OSX
On the Pi's side i call write with baudrate of ```57600``` and a packet size of ```57600```
It works great even though it seems quite slow to write.

Now on the other side my parser function get called with buffers of size 1 to 10 each time. I finally get the whole package, but it makes the process very slow.

Is there a way to make read packages bigger? like close the baudrate?

Thanks

reconbot, 2017-06-20:
There is not a way to do it directly. Data is emitted as it is received. You can however concatenate the data you receive or use the bytelength parser (in 5x) which does it for you.

closed 2017-06-20 by reconbot

farfromrefug, 2017-06-20:
@reconbot i have seen that. But don't you think it would be much more efficient to write parsers in C then so that we concatenate the buffers in C and not in JS.
I think i will try test this in my work to see if it makes a difference with big packages

reconbot, 2017-06-20:
I think it really depends on the use case. It's not uncommon to keep a
larger buffer around to fill and slice data off of as needed to save on
memory allocations. You could write a c++ binding that only emits data
after it hits a size or timeout but that would add latency to receiving the
data. You'd want to see if the performance of buffer handling matters with
the introduced latency.

---
Francis Gulotta
wizard@roborooter.com

farfromrefug, 2017-06-20:
@reconbot what i am saying is that emiting for every single byte and handling it on the JS side could be really slow.
I will try

reconbot, 2017-06-20:
We emit as it's loaded, not every byte. If more data is flushed from the OS
at a time, that's what you'll get. We ask for up to... 2k at a time? It's
probably slightly different per version of serialport.

---
Francis Gulotta
wizard@roborooter.com

farfromrefug, 2017-06-20:
@reconbot in my version 4.0.1 i literally get between 1 and 3 bytes each time! Thus that issue