serialport/node-serialport #1129: SIGSEGV, Segmentation fault using 5.0.0-beta3

andrei-yanovich, 2017-04-04:
- __SerialPort Version__: 5.0.0-beta3
- __NodeJS Version__: v6.10.1
- __Operating System and Hardware Platform__: Docker image based on Ubuntu running on Ubuntu both 14.04

- Have you checked the right __version__ of the __api docs?__: Yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? No
- Are you using __Electron__ and have you checked the _Electron Docs_?: No

### Summary of Problem
I randomly get SIGSEGV exception. I have a small script with a line parser and byte parser.  Line parser logs everything from serial.
Byte parser is used for finding a specific substring (WaitFor method). The script just sends several commands and after is waiting for some string appear from serial.

### Steps and Code to Reproduce the Issue

backtrace from core dump:
`
#0  0x00007faf53c0caba in v8::PersistentBase<v8::Function>::IsEmpty (this=0x5b70) at /root/.node-gyp/6.10.1/include/node/v8.h:521
521	  V8_INLINE bool IsEmpty() const { return val_ == NULL; }
(gdb) bt
#0  0x00007faf53c0caba in v8::PersistentBase<v8::Function>::IsEmpty (this=0x5b70) at /root/.node-gyp/6.10.1/include/node/v8.h:521
#1  0x00007faf53c0d36a in v8::PersistentBase<v8::Function>::Reset (this=0x5b70) at /root/.node-gyp/6.10.1/include/node/v8.h:7607
#2  0x00007faf53c0ca56 in Nan::Persistent<v8::Function, v8::NonCopyablePersistentTraits<v8::Function> >::Reset (this=0x5b70)
    at ../../nan/nan_persistent_12_inl.h:25
#3  0x00007faf53c0b5be in Nan::Callback::~Callback (this=0x5b70, __in_chrg=<optimized out>) at ../../nan/nan.h:1377
#4  0x00007faf53c1008d in ReadPoller::_close (this=0x31fd120) at ../src/read-poller.cpp:100
#5  0x00007faf53c0fd3a in _serialportReadable (req=0x31fd138, status=0, events=1) at ../src/read-poller.cpp:40
#6  0x000000000132b990 in uv__io_poll (loop=loop@entry=0x1e1c980 <default_loop_struct>, timeout=-1)
    at ../deps/uv/src/unix/linux-core.c:380
#7  0x000000000131bea6 in uv_run (loop=0x1e1c980 <default_loop_struct>, mode=UV_RUN_ONCE) at ../deps/uv/src/unix/core.c:354
#8  0x00000000010a5ae0 in node::Start(int, char**) ()
#9  0x00007faf55e3cf45 in __libc_start_main (main=0x7b7070 <main>, argc=4, argv=0x7ffec2f80208, init=<optimized out>,
    fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7ffec2f801f8) at libc-start.c:287
#10 0x00000000007b877d in _start ()
`
Script:
`
'use strict';

const colors = require("colors");
const fs  = require("fs");
const yargs = require("yargs");
const SerialPort = require('serialport');
const Parsers = SerialPort.parsers;

const argv = require('yargs')
    .usage('Usage: $0 -p <serial_port_name>')
    .example('$0 -p /dev/ttyUSB0', '')
    .alias('p', 'port')
    .nargs('p', 1)
    .demand('p', 1)
    .describe('p', 'Serial port name')
    .help('h')
    .alias('h', 'help')
    .argv;

const serialPort = argv.p;

console.log("INFO:".blue + " Opening serial port [" + serialPort + "]".white);

const port = new SerialPort(serialPort, { baudRate: 115200 });
port.on('open', function() {
    console.log("INFO:".blue + " Serial port open");
    Start();
});

port.on('error', function(error) {
    console.log("Error:".red + " General serial port error [".white + error + "]".white);
});

const lineParser = new Parsers.Readline({ delimiter: "\n" });
port.pipe(lineParser);
lineParser.on("data", function(data) {
    console.log(data.blue);
});

// Create parser, pipe to port and setup reception
const byteParser = new Parsers.ByteLength({ length:1 });
port.pipe(byteParser);

let dataHandler;
byteParser.on('data', function(byte) {
    if (dataHandler) {
        dataHandler(byte);
    }
});

function Start() {
    WaitFor('Booting Bootloader...')
    .then(() => Send([3,3,3,3,3,3,13,10]))
    .then(() => WaitFor('SSBL>'))
    .then(() => Send('debug'))
    .then(() => WaitFor('DEBUG>'))
    .then(() => WaitFor('FOREVER!!!!!!'))
}

function WaitFor(...waitStr) {
    let textFinders = [];

    return new Promise((resolve, reject) => {
            function MatchCallback(text) {
                dataHandler = null;
                resolve(text);
                console.log("WAITFOR:".bgWhite.bold.black + " " + "(MATCH) -> ".green
                    + text.red);
            }

            function Handler(byte) {
                for (let i = 0; i < textFinders.length; i++) {
                    textFinders[i].PushByte(byte);
                }
            }

            console.log("WAITFOR:".bgWhite.bold.black + " " + "(WAIT) -> ".yellow
                + JSON.stringify(waitStr).red);

            // Create textfinder objects
            for (let i = 0; i < waitStr.length; i++)
            {
                textFinders[i] = new TextFinder(waitStr[i], MatchCallback);
            }

            dataHandler = Handler;
        }
    );
}

function Send(data) {
    return new Promise(
        function(resolve)
        {
            port.write(data, function()
            {
                resolve()
            });
        }
    );
}

function TextFinder(text, onFind) {
    let findPos = 0;
    const waitStr = text;

    const PushByte = (byte) => {
        // If received byte doesn't match pattern, reset search
        if (byte != waitStr[findPos]) {
            findPos = 0;
            return;
        }

        findPos++;

        // String match: invoke callback
        if (findPos >= waitStr.length) {
            if (onFind) {
                onFind(waitStr);
            }
        }
    };

    return { PushByte };
}
`

reconbot, 2017-04-04:
Thank you this is helpful.

andrei-yanovich, 2017-04-04:
@reconbot, 
Do you have an idea what the root cause is and how it can be fixed?

reconbot, 2017-06-20:
Can you try with beta5? I cleaned up some of the writing code.

closed 2017-07-21 by reconbot