serialport/node-serialport #1142: Error while install serial port on rpi3

bidyutper, 2017-04-11:
$ npm install serialport --save
npm WARN package.json http@0.0.0 No description
npm WARN package.json http@0.0.0 No repository field.
npm WARN package.json http@0.0.0 No README data
npm WARN package.json mraa@1.6.1 No repository field.
npm ERR! Error: EACCES, symlink '../serialport/bin/serialport-list.js'
npm ERR!  { [Error: EACCES, symlink '../serialport/bin/serialport-list.js']
npm ERR!   errno: 3,
npm ERR!   code: 'EACCES',
npm ERR!   path: '../serialport/bin/serialport-list.js' }
npm ERR!
npm ERR! Please try running this command again as root/Administrator.

npm ERR! System Linux 4.4.38-v7+
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install" "serialport" "--save"
npm ERR! cwd /home/pi
npm ERR! node -v v0.10.29
npm ERR! npm -v 1.4.21
npm ERR! path ../serialport/bin/serialport-list.js
npm ERR! code EACCES
npm ERR! errno 3
npm ERR! stack Error: EACCES, symlink '../serialport/bin/serialport-list.js'
npm ERR!
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/npm-debug.log
npm ERR! not ok code 0

bidyutper, 2017-04-11:
in root
:/home/pinpm install serialport --save                   npm WARN package.json http@0.0.0 No description
npm WARN package.json http@0.0.0 No repository field.
npm WARN package.json http@0.0.0 No README data
npm WARN package.json mraa@1.6.1 No repository field.
|
> serialport@4.0.7 install /home/pi/node_modules/serialport
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v11-linux-arm.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and node@0.10.29 (node-v11 ABI) (falling back to source compile with node-gyp)
make: Entering directory '/home/pi/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
In file included from ../src/./serialport.h:7:0,
                 from ../src/serialport.cpp:1:
../node_modules/nan/nan.h:325:47: error: ‘REPLACE_INVALID_UTF8’ is not a member of ‘v8::String’
   static const unsigned kReplaceInvalidUtf8 = v8::String::REPLACE_INVALID_UTF8;
                                               ^
serialport.target.mk:84: recipe for target 'Release/obj.target/serialport/src/serialport.o' failed
make: *** [Release/obj.target/serialport/src/serialport.o] Error 1
make: Leaving directory '/home/pi/node_modules/serialport/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/share/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:809:12)
gyp ERR! System Linux 4.4.38-v7+
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "build" "--fallback-to-build" "--module=/home/pi/node_modules/serialport/build/Release/serialport.node" "--module_name=serialport" "--module_path=/home/pi/node_modules/serialport/build/Release"
gyp ERR! cwd /home/pi/node_modules/serialport
gyp ERR! node -v v0.10.29
gyp ERR! node-gyp -v v0.12.2
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute 'node-gyp build --fallback-to-build --module=/home/pi/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/home/pi/node_modules/serialport/build/Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/pi/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
node-pre-gyp ERR! stack     at maybeClose (child_process.js:755:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:822:5)
node-pre-gyp ERR! System Linux 4.4.38-v7+
node-pre-gyp ERR! command "node" "/home/pi/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/pi/node_modules/serialport
node-pre-gyp ERR! node -v v0.10.29
node-pre-gyp ERR! node-pre-gyp -v v0.6.32
node-pre-gyp ERR! not ok
Failed to execute 'node-gyp build --fallback-to-build --module=/home/pi/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/home/pi/node_modules/serialport/build/Release' (1)
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian

npm ERR! serialport@4.0.7 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@4.0.7 install script.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! System Linux 4.4.38-v7+
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install" "serialport" "--save"
npm ERR! cwd /home/pi
npm ERR! node -v v0.10.29
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm ERR!
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/npm-debug.log
npm ERR! not ok code 0

fivdi, 2017-04-13:
It's difficult to tell why the `EACCES` error mentioned in the first post is occurring without more background information. It looks like `npm install serialport --save` was called in a directory where root privileges are needed to perform the install.

The `‘REPLACE_INVALID_UTF8’ is not a member of ‘v8::String’` error in the second post is occurring because the version of Node.js that comes pre-installed with Raspbian, Node.js v0.10.29, is broken. For further information related to this specific issue see https://github.com/fivdi/onoff/wiki/Node.js-v0.10.29-and-native-addons-on-the-Raspberry-Pi.

The best thing to do is to install an up to date version of Node.js on the Raspberry Pi and take it from there. Also, don't use `sudo` when installing packages unless there is a good reason for doing so.

There is additional information in the [readme](https://github.com/EmergingTechnologyAdvisors/node-serialport#raspberry-pi-linux) related to installing serialport on the Raspberry Pi and installing with root privileges.

closed 2017-06-20 by reconbot