serialport/node-serialport #1143: Can't catch error on `new SerialPort(...)`

jacobq, 2017-04-11:
- __SerialPort Version__: 4.0.7
- __NodeJS Version__: 6.10.2
- __Operating System__ and __Hardware Platform__: Windows 10 Pro, x86_64

### Summary of Problem
Some COM devices generate errors when attempting to connect (in any app, not just with `node-serialport`). For example, when attempting to open an unconfigured [HHD Virtual Serial Port](https://www.hhdsoftware.com/virtual-serial-ports), PuTTY shows an error like "Error reading from serial device". When one attempts to access a device like this with `node-serialport` it causes a fatal error. It's important to be able to catch this error (especially since it doesn't appear possible to tell in advance if calling the constructor will cause it to happen or not) to prevent it from crashing an entire application.

### Steps and Code to Reproduce the Issue

Run `npm install serialport` in an empty directory then run the following in the node REPL:
```
try {
    var SP = require('serialport');
    new SP('COM1', {}, function() {
        console.log('OK', arguments); 
    });
} catch(e) {
    console.log('ERROR', e)
}
```

Sample output from running above:
```
SerialPort {
  domain:
   Domain {
     domain: null,
     _events: { error: [Function] },
     _eventsCount: 1,
     _maxListeners: undefined,
     members: [] },
  _events: {},
  _eventsCount: 0,
  _maxListeners: undefined,
  path: 'COM1',
  fd: null,
  paused: true,
  opening: false,
  closing: false,
  options:
   { baudRate: 9600,
     autoOpen: true,
     parity: 'none',
     xon: false,
     xoff: false,
     xany: false,
     rtscts: false,
     hupcl: true,
     dataBits: 8,
     stopBits: 1,
     bufferSize: 65536,
     lock: true,
     parser: [Function: raw],
     platformOptions: {},
     disconnectedCallback: [Function: bound ],
     dataCallback: [Function: bound raw] } }
> OK { '0': null }
undefined:0

TypeError: undefined is not a function
```

reconbot, 2017-06-20:
If you disable `autoOpen` and then use the open callback what happens?

jacobq, 2017-06-20:
@reconbot I'm currently swamped with some other work but will give this a try and let you know by next week. Thanks for the suggestion.

reconbot, 2017-07-10:
Any word on this? I'm pushing for a 5.0.0 release and probably wont do any more maintenance on 4.0.0 if I can help it. In any case I'm going to close this issue for now, but if you'd like to continue with it feel free to comment and we'll reopen.

jacobq, 2017-07-10:
@reconbot Sorry for letting this fall off my radar. 

When I run the test program below I get the output shown:
```
console.log('1 - require');
var SerialPort = require('serialport');
console.log('2 - new');
var port = new SerialPort('COM5', {autoOpen: false});
console.log('3 - waiting');
setTimeout(function() {
  console.log('4 - trying open');
  try {
    port.open(function (err) {
      if (err) {
        console.log('6a - Error opening port: ', err.message);
      }
      else {
        console.log('6b - successfully opened port');
      }
    });
    console.log('5 - finished calling open');
  }
  catch (e) {
    console.log('7 - caught', e);
  }
  finally {
    console.log('8 - finally');
  }

  // the open event will always be emitted
  port.on('open', function() {
    console.log('9 - handling open event');
  });
}, 1000);
```

```
1 - require
2 - new
3 - waiting
4 - trying open
5 - finished calling open
8 - finally
9 - handling open event
6b - successfully opened port
undefined:0

TypeError: undefined is not a function

```

reconbot, 2017-07-10:
There are no sync errors other than argument errors so you shouldn't need any try catches, they don't work for callbacks.

What version did you test?

jacobq, 2017-07-10:
node v6.11.0 & serialport 4.0.7

jacobq, 2017-07-10:
Where is the `TypeError` coming from? Is there no way that I can wrap a try/catch around it? I am not really concerned about "fixing" this problem. I just need a way to handle it gracefully (i.e. not let it crash the entire process).

reconbot, 2017-07-10:
Without a stack trace I can't really tell you. Does this happen without the `setTimeout`? Does it happen with `serialport@5.0.0-beta6`?

jacobq, 2017-07-10:
I will try both, but I highly suspect the answer is yes. There seems to be some assumption in the underlying code for the Windows platform that doesn't hold for disconnected / not-ready COM ports.

reconbot, 2017-07-11:
Our integration tests do run against bad comports. I haven't been able to
reproduce your error.

jacobq, 2017-07-11:
@reconbot is integration testing your only Windows environment? I am able to reproduce this very consistently in vmware + Windows 10 using [a "free" virtual serial port utility](https://freevirtualserialports.com/). It is not even necessary to purchase a Windows license as you can use a [free virtual machine images provided Microsoft](https://developer.microsoft.com/en-us/microsoft-edge/tools/vms/). If it would be helpful to you I can configure one with node, a virtual serial port, and a sample program demonstrating the problem.

reconbot, 2017-07-16:
I test on vms with real hardware attached. The tests will run on any computer. `set TEST_PORT=COM3; npm test` (for com3 for example)

With node8 unhandled error events give the promise rejection warning. `port.on('error', cb)` will handle that. Write errors aren't always able to give the error in the callback.

From the [stream docs](https://nodejs.org/api/stream.html#stream_writable_write_chunk_encoding_callback)
> If an error occurs, the callback may or may not be called with the error as its first argument. To reliably detect write errors, add a listener for the 'error' event.

As for your write error on the virtual com port.
> `(node:5548) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 1): Error: Writing to COM port (GetOverlappedResult): Unknown error code 87`

That error code is probably coming from your driver. They might know what it means. The code you want to share with is [here](https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/master/src/serialport_win.cpp#L284-L324) it's possible they require a setting that you're not setting or something. I know people have had success with virtual com ports.

reconbot, 2017-09-11:
I'd love to hear if the virtual serialport people had any thoughts. For now I'm going to close this issue to due age, but if you'd like to continue with it feel free to comment and we'll reopen.

closed 2017-09-11 by reconbot