serialport/node-serialport #1145: a error when I pulled up CH340

hanfengcan, 2017-04-13:
- __SerialPort Version__: master
- __NodeJS Version__: 7.8.0
- __Operating System__ and __Hardware Platform__: win10 & ch340

### Summary of Problem
> (node:13524) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 1): Error: Reading from COM port (ReadFile): Access denied
(node:13524) DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
### Steps and Code to Reproduce the Issue
everything is ok, but I pulled up the CH340(usb to TTL) when program is running.

this is my code
```
/* eslint-disable require-path-exists/exists */
'use strict';

// Use a Readline parser

const SerialPort = require('serialport');
const parsers = SerialPort.parsers;

// Use a `\r\n` as a line terminator
const parser = new parsers.Readline({
  delimiter: '\r\n'
});

const port = new SerialPort('COM9', {
  baudRate: 57600
});

port.pipe(parser);

port.on('open', () => console.log('Port open'));

parser.on('data', console.log);

port.write('ROBOT PLEASE RESPOND\n');

port.on('disconnect', ()=>{
    console.log('disconnect');
});

port.on('close', ()=>{
    console.log('close');
});

port.on('error', (err)=>{
    console.log('error: ', err.message);
})
// The parser will emit any string response
```

reconbot, 2017-06-20:
Please try the latest beta. 

I'm going to close this issue due to it's age, but if you'd like to continue with it feel free to comment and we'll reopen.

hanfengcan, 2017-06-20:
OK @reconbot

hanfengcan, 2017-06-28:
@reconbot 
Still has this problem when I pull out the CH340 suddenly
```
(node:15104) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 1): Error: Reading from COM port (ReadFile): Access denied
(node:15104) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

hanfengcan, 2017-06-30:
@reconbot I found my disconnect event callback don't work when I disconnect CH340 before I close port
```
port.on('disconnect', () => {
  console.log('disconnect')
});
```

reconbot, 2017-07-05:
Sorry to have this issue closed. Please try `5.0.0-beta6` I reworked how disconnects happen.

hanfengcan, 2017-07-06:
great :smiley:

reconbot, 2017-07-06:
closed via #1217 =)

closed 2017-07-06 by reconbot