serialport/node-serialport #1146: node-pre-gyp: not found. If root project path contains 'node_modules'

atamanuk, 2017-04-14:
- __SerialPort Version__: 5.0.0-beta1 and 4.0.7
- __NodeJS Version__: 5.8.0
- __Operating System__ and __Hardware Platform__: Ubuntu 15.10 x86_64

- Have you checked the right __version__ of the __api docs?__: Yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? Yes
- Are you using __Electron__ and have you checked the _Electron Docs_?: No

### Summary of Problem
The library is not installed when the path to root project directory contains the word 'node_modules'.

```
➜  test_node_modules npm i

> serialport@4.0.7 install /home/xxxxx/test_node_modules/node_modules/serialport
> node-pre-gyp install --fallback-to-build

sh: 1: node-pre-gyp: not found
npm WARN test_serial_port@1.0.0 No repository field.
npm WARN test_serial_port@1.0.0 No license field.
npm ERR! Linux 4.2.0-42-generic
npm ERR! argv "/home/xxxxx/.nvm/v5.8.0/bin/node" "/home/xxxxx/.nvm/v5.8.0/bin/npm" "i"
npm ERR! node v5.8.0
npm ERR! npm  v3.7.3
npm ERR! file sh
npm ERR! code ELIFECYCLE
npm ERR! errno ENOENT
npm ERR! syscall spawn

npm ERR! serialport@4.0.7 install: `node-pre-gyp install --fallback-to-build`
npm ERR! spawn ENOENT
npm ERR! 
npm ERR! Failed at the serialport@4.0.7 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs serialport
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/xxxxx/test_node_modules/npm-debug.log
```

package.json
```
{
  "name": "test_serial_port",
  "description": "test",
  "version": "1.0.0",
  "scripts": {
  },
  "licenses": "UPL-1.0",
  "main": "index.js",
  "dependencies": {
    "serialport": "4.0.7"
  }
}
```

### Steps and Code to Reproduce the Issue
I renamed the directory to "test_node_modullles" :)

ljubomirb, 2017-06-01:
This is strange, after a few hours of trying everything to cross compile serial port, I tried this approach, and it worked!? Don't know if anything else was changed, since I tried a lot, but after changing the name of node_modules in:
`(...)source/build_dir/target-mips_24kc_musl/node-serialport-4.0.7/ipkg-install/usr/lib` during cross compilation for LEDE project, everything went well. Before this, I was unable to compile other packages like sqlite or similar.

reconbot, 2017-06-20:
Glad you figured it out. Nodejs does look for node_modules in the parent folders up to the root folder.

closed 2017-06-20 by reconbot