serialport/node-serialport #1149: node-pre-gyp install --fallback-to-build

MelissaPereira, 2017-04-17:
- __SerialPort Version__: 4.0.7
- __NodeJS Version__: 6.10.2
- __Operating System__ and __Hardware Platform__: Debian 8.6 (jessie)

### Summary of Problem
I get errors when trying to install serialport. It does install but with errors and when I try to run a snippet of code, I get the error that there is no such directory /dev/tty-usbserial1

### Steps and Code to Reproduce the Issue
I ran npm i serialport to install it first and got this

```
root@beaglebone:~/assignment2-team87/Task 2# npm i serialport

> roboticscape@0.0.2 install /root/assignment2-team87/Task 2/node_modules/roboticscape
> node-gyp rebuild

make: Entering directory '/root/assignment2-team87/Task 2/node_modules/roboticscape/build'
  CXX(target) Release/obj.target/roboticscape/rc-node-bindings.o
../rc-node-bindings.cc:8:30: fatal error: roboticscape.h: No such file or directory
     #include <roboticscape.h>
                              ^
compilation terminated.
roboticscape.target.mk:94: recipe for target 'Release/obj.target/roboticscape/rc-node-bindings.o' failed
make: *** [Release/obj.target/roboticscape/rc-node-bindings.o] Error 1
make: Leaving directory '/root/assignment2-team87/Task 2/node_modules/roboticscape/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Linux 4.4.30-ti-r64
gyp ERR! command "/usr/bin/nodejs" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /root/assignment2-team87/Task 2/node_modules/roboticscape
gyp ERR! node -v v6.10.2
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok

> serialport@4.0.7 install /root/assignment2-team87/Task 2/node_modules/serialport
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v48-linux-arm.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and node@6.10.2 (node-v48 ABI) (falling back to source compile with node-gyp)
make: Entering directory '/root/assignment2-team87/Task 2/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
../src/serialport.cpp: In function 'int getIntFromObject(v8::Local<v8::Object>, std::string)':
../src/serialport.cpp:90:52: warning: 'v8::Local<v8::Int32> v8::Value::ToInt32() const' is deprecated (declared at /root/.node-gyp/6.10.2/include/node/v8.h:8202): Use maybe version [-Wdeprecated-declarations]
   return getValueFromObject(options, key)->ToInt32()->Int32Value();
                                                    ^
../src/serialport.cpp: In function 'double getDoubleFromObject(v8::Local<v8::Object>, std::string)':
../src/serialport.cpp:102:53: warning: 'v8::Local<v8::Number> v8::Value::ToNumber() const' is deprecated (declared at /root/.node-gyp/6.10.2/include/node/v8.h:8166): Use maybe version [-Wdeprecated-declarations]
   return getValueFromObject(options, key)->ToNumber()->NumberValue();

```

```
var SerialPort = require("serialport");
var port = new SerialPort("/dev/tty-usbserial1", {
  baudRate: 57600
});
```
then when I run the snippet of code from the npm page shown above I get this

```
root@beaglebone:~/assignment2-team87/Task 2# node test.js
events.js:160
      throw er; // Unhandled 'error' event
      ^

Error: Error: No such file or directory, cannot open /dev/tty-usbserial1
    at Error (native)

```

fivdi, 2017-04-17:
As far as I can tell there is only one "real" error shown above and that's:
```
make: Entering directory '/root/assignment2-team87/Task 2/node_modules/roboticscape/build'
  CXX(target) Release/obj.target/roboticscape/rc-node-bindings.o
../rc-node-bindings.cc:8:30: fatal error: roboticscape.h: No such file or directory
     #include <roboticscape.h>
                              ^
compilation terminated.
```
This error is related to `roboticscape` which is a dependency of `bonescript`. It's not clear to me why calling `npm i serialport` results in npm attempting to reinstall `roboticscape`. Irrespective of why this is happening, I don't think it has anything to do with `serialport`.

After failing to install `robiticscape`, `serialport` is installed and the following output is displayed:

```
> serialport@4.0.7 install /root/assignment2-team87/Task 2/node_modules/serialport
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v48-linux-arm.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and node@6.10.2 (node-v48 ABI) (falling back to source compile with node-gyp)
make: Entering directory '/root/assignment2-team87/Task 2/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
../src/serialport.cpp: In function 'int getIntFromObject(v8::Local<v8::Object>, std::string)':
../src/serialport.cpp:90:52: warning: 'v8::Local<v8::Int32> v8::Value::ToInt32() const' is deprecated (declared at /root/.node-gyp/6.10.2/include/node/v8.h:8202): Use maybe version [-Wdeprecated-declarations]
   return getValueFromObject(options, key)->ToInt32()->Int32Value();
                                                    ^
../src/serialport.cpp: In function 'double getDoubleFromObject(v8::Local<v8::Object>, std::string)':
../src/serialport.cpp:102:53: warning: 'v8::Local<v8::Number> v8::Value::ToNumber() const' is deprecated (declared at /root/.node-gyp/6.10.2/include/node/v8.h:8166): Use maybe version [-Wdeprecated-declarations]
   return getValueFromObject(options, key)->ToNumber()->NumberValue();
```

Although the text `node-pre-gyp ERR` is displayed, there isn't actually an error. When `npm i serialport` is called, an attempt is made to download pre-compiled `serialport` binaries for the BeagleBone but none are available. Because none are available the `node-pre-gyp ERR` is displayed. As a fallback, the source code for `serialport` is compiled on the BeagleBone. During this compilation a number of warnings are displayed that can be safely ignored.

As you mention, `serialport` was installed successfully.

The snippet of code being executed throws an error because the file `/dev/tty-usbserial1` doesn't exist. Check this with the command `ls /dev/tty-usbserial1`.

You'll need to figure out which file in `/dev` corresponds to the serial port that you would like to use and pass the name of that file to the `SerialPort` constructor.

MelissaPereira, 2017-04-18:
I'll look into that :) Thank you for the help!

closed 2017-06-20 by reconbot