serialport/node-serialport #1151: npm install serial-port fails on Beaglebone

billtrudell, 2017-04-18:
- __SerialPort Version__:
- __NodeJS Version__:
- __Operating System__ and __Hardware Platform__:

- Have you checked the right __version__ of the __api docs?__:
- Are you having trouble installing and you checked the __Installation Special Cases__ docs?
- Are you using __Electron__ and have you checked the _Electron Docs_?:

### Summary of Problem
root@beaglebone:/home/Node_Serial# npm install serial-port
npm ERR! Linux 4.4.9-ti-r25
npm ERR! argv "/root/nodejs/bin/node" "/usr/local/bin/npm" "install" "serial-port"
npm ERR! node v7.8.0
npm ERR! npm  v4.2.0
npm ERR! code ENOTFOUND
npm ERR! errno ENOTFOUND
npm ERR! syscall getaddrinfo

npm ERR! network getaddrinfo ENOTFOUND registry.npmjs.org registry.npmjs.org:443
npm ERR! network This is most likely not a problem with npm itself
npm ERR! network and is related to network connectivity.
npm ERR! network In most cases you are behind a proxy or have bad network settings.
npm ERR! network
npm ERR! network If you are behind a proxy, please make sure that the
npm ERR! network 'proxy' config is set properly.  See: 'npm help config'

npm ERR! Please include the following file with any support request:
npm ERR!     /root/.npm/_logs/2016-05-13T19_32_49_096Z-debug.log

### Steps and Code to Reproduce the Issue
Just execute "npm install serial-port"...

[serial-port-error.txt](https://github.com/EmergingTechnologyAdvisors/node-serialport/files/931182/serial-port-error.txt)

fivdi, 2017-04-19:
There's a typo in the install command. It should be `npm install serialport` rather than `npm install serial-port`.

The BeagleBone appears to have network connectivity issues as it doesn't appear to able to communicate with the npm registry.

The BeagleBone also appears to have date/time issues as one of the error messages references a file called `/root/.npm/_logs/2016-05-13T19_32_49_096Z-debug.log`. The timestamp on the file is from May 2016.

billtrudell, 2017-04-20:
Good catch.  I'm now getting this error, leading me to think serialport is not supported on the versions of node I am running:

13987 error Linux 4.4.9-ti-r25
13988 error argv "/root/nodejs/bin/node" "/usr/local/bin/npm" "install" "./node-serialport-master/"
13989 error node v7.8.0
13990 error npm  v4.2.0
13991 error code ELIFECYCLE
13992 error errno 126
13993 error serialport@5.0.0-beta3 install: `node-pre-gyp install --fallback-to-build`
13993 error Exit status 126
13994 error Failed at the serialport@5.0.0-beta3 install script 'node-pre-gyp install --fallback-to-build'.

closed 2017-04-20 by billtrudell

fivdi, 2017-04-20:
Please post the exact command used to install serialport and the entire output of that command.

billtrudell, 2017-04-20:
Since I had firewall issues, I downloaded a zip and did a local install as follows:

root@beaglebone:/home/Node_Serial# sudo npm install ./node-serialport-master/

> serialport@5.0.0-beta3 install /home/Node_Serial/node_modules/serialport
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download(undefined): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/5.0.0-beta3/serialport-v5.0.0-beta3-node-v51-linux-arm.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for serialport@5.0.0-beta3 and node@7.8.0 (node-v51 ABI) (falling back to source compile with node-gyp)
gyp WARN install got an error, rolling back install
gyp ERR! configure error
gyp ERR! stack Error: read ECONNRESET
gyp ERR! stack     at exports._errnoException (util.js:1034:11)
gyp ERR! stack     at TLSWrap.onread (net.js:580:26)
gyp ERR! System Linux 4.4.9-ti-r25
gyp ERR! command "/root/nodejs/bin/node" "/root/nodejs/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "configure" "--fallback-to-build" "--module=/home/Node_Serial/node_modules/serialport/build/Release/serialport.node" "--module_name=serialport" "--module_path=/home/Node_Serial/node_modules/serialport/build/Release"
gyp ERR! cwd /home/Node_Serial/node_modules/serialport
gyp ERR! node -v v7.8.0
gyp ERR! node-gyp -v v3.5.0
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute '/root/nodejs/bin/node /root/nodejs/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js configure --fallback-to-build --module=/home/Node_Serial/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/home/Node_Serial/node_modules/serialport/build/Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/Node_Serial/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:106:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:194:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:899:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
node-pre-gyp ERR! System Linux 4.4.9-ti-r25
node-pre-gyp ERR! command "/root/nodejs/bin/node" "/home/Node_Serial/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/Node_Serial/node_modules/serialport
node-pre-gyp ERR! node -v v7.8.0
node-pre-gyp ERR! node-pre-gyp -v v0.6.34
node-pre-gyp ERR! not ok
Failed to execute '/root/nodejs/bin/node /root/nodejs/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js configure --fallback-to-build --module=/home/Node_Serial/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/home/Node_Serial/node_modules/serialport/build/Release' (1)
npm WARN node_serial@1.0.0 No description
npm WARN node_serial@1.0.0 No repository field.
npm ERR! Linux 4.4.9-ti-r25
npm ERR! argv "/root/nodejs/bin/node" "/usr/local/bin/npm" "install" "./node-serialport-master/"
npm ERR! node v7.8.0
npm ERR! npm  v4.2.0
npm ERR! code ELIFECYCLE
npm ERR! errno 1

npm ERR! serialport@5.0.0-beta3 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@5.0.0-beta3 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs serialport
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

fivdi, 2017-04-20:
The ECONNRESET indicates that there are still connectivity issues.

Unfortunately downloading the zip file for serialport isn't enough as serialport has [several dependencies](https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/master/package.json#L55-L61) that also need to be installed. These dependencies in turn will have their depencencies which will also have to be installed. Even though the code for serialport has been copied onto the BeagleBone, all the dependencies still have to be installed together with serialport.

This approach isn't going to be practical.

You'll need an internet connection for the install to function correctly.

billtrudell, 2017-04-20:
I understand.  However, convincing companies to allow unrestricted access to the Internet for unknown resources (dependencies) isn't safe...  This is incredibly problematic, it's one roadblock after another to add IP addresses and sites to exception lists on a firewall.  I can't tell them what sites any given execution of "npm install...." will need.

susacha, 2017-06-04:
@billtrudell 
Did you find a resolution to this problem? I also have machines that I want to use node.js on with the serialport install but those machines don't have internet access and aren't allowed to have it.

fivdi, 2017-06-04:
@susacha The following may work: flash an SD card with Debian, stick it in a BeagleBone Black that does have Internet access and installing serialport. This SD card could then be duplicated for each BeagleBone Black that doesn't have Internet access.

billtrudell, 2017-06-04:
I was unable to do it locally, I had my IT department grant access to the following sites:  https://nodejs.org
http://github.com
http://registry.npmjs.org

susacha, 2017-06-04:
@fivdi Thanks for the suggestion but my problem is only similar to billtrudell. I am not using a BeagleBone. I have a Numato gpio board that I am writing a program that needs to read info from the com port via IE. I wanted to use the Node Serialport Library to do that.