serialport/node-serialport #1153: Missing ports in SerialPort.list

fly19890211, 2017-04-21:
- __SerialPort Version__: 4.0.7
- __NodeJS Version__: 4.4.7
- __Operating System__ and __Hardware Platform__: ubuntu 14.04

- Have you checked the right __version__ of the __api docs?__: yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? nope
- Are you using __Electron__ and have you checked the _Electron Docs_?: nope, nope

### Summary of Problem
Hello everyone,
i have a machine as datalogger manufactured by Adventech and and expansion RS485 card is install. i'm able to use those ports as /dev/ttyAP0, /dev/ttyAP1. but i can't find it by SerialPort.list
the screenshot:
http://imgur.com/a/RvNwt
thx
### Steps and Code to Reproduce the Issue

reconbot, 2017-06-20:
We do filter these ports out. I've never heard of AP tty subsystem. I can't find any docs on it however. 

Here's the code responsible. https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/4x/lib/list-unix.js#L80

reconbot, 2017-07-05:
If you can find me more information about the AP TTY subsystem I'll be happy to add it to the list.

closed 2017-07-15 by reconbot

reconbot, 2017-07-17:
This will be fixed in `5.0.0-beta8`

PR #1234 (2017-07-15): [linux] Add the `ttyAP` subsystem to serialport list
Closes #1153