serialport/node-serialport #1159: list function always return empty list.

yantrab, 2017-04-28:
- __SerialPort Version__: 4.0.7
- __NodeJS Version__:v7.7.1
- __Operating System__ and __Hardware Platform__:windows 10 

### Summary of Problem
Hi, i try to mange the ports, part the list function always return empty list.
 
### Steps and Code to Reproduce the Issue

C:\>serialport-list  -f json
[]

And in nodejs:

` SerialPort.list((err, ports) => {})`

In my c# project it work fine, with GetPortNames function.

hanfengcan, 2017-06-02:
test this code
```
serialport.list((err, ports) => {
    ports.forEach((port) => {
      comList.push(port)
    })
  })
```

reconbot, 2017-06-20:
Serialport.list is asynchronous so that example code wont work.

Are you sure the devices have com ports?

closed 2017-07-05 by reconbot