serialport/node-serialport #1161: TypeError: First argument must be an int at Promise using Parsers under Electron on Win10x64

ArtemKosogorov, 2017-05-03:
- SerialPort Version **5.0.0-beta3**:
- NodeJS Version **7.4.0** inside **Elecron 1.6.6** (also tried in Electron 1.6.2, 1.4.7)
- Operating System: Windows10 and Intel x64:
 (Checked on OsX Sierra 10.12.4 - same code works without this error, still having parser problems).

- Have you checked the right version of the eapi docs?: **yes**  (5.0.0)
- Are you having trouble installing and you checked the docs? **yes**
- Are you using Electron and have you checked the Electron Docs?: **yes**
Tried: `npm rebuild --build-from-source ` and `electron-rebuild` - no result after rebuild.

### Summary of Problem
Opening a Serialport instance with any parser attached invokes the TypeError -  **Uncaught (in promise) TypeError: First argument must be an int.** 
Port is opened and .write works, yet parser doesn`t function.

Full error message:
```
…\node_modules\serialport\lib\util.js:13 
Uncaught (in promise) TypeError: First argument must be an int
    at Promise (…\node_modules\serialport\lib\util.js:13:12)
    at …\node_modules\serialport\lib\util.js:5:12
    at read.then (…\node_modules\serialport\**lib\bindings\win32.js**:39:42)
```
### Attempts to fix issue: 
1. Tried: `npm rebuild --build-from-source ` and `electron-rebuild` - no result after rebuild.
2. Excluding parser attachment .pipe function - works OK (yet no reading), 
3. Replacing parser type (tried Readline and Delimiter)
4. Attaching parser .on callback to port instead of parser
4. Changing platform - same code works on OsX Sierra 10.12.4
Question: Is it OK that under Win64 the `win32.js` bindings file is involved?

### Steps and Code to Reproduce the Issue

```
var SerialPort = require('serialport');
var Readline = SerialPort.parsers.Readline;
var port = new SerialPort('COM4',{ autoOpen: true });
var parser = port.pipe(new Readline());
port.pipe(parser);
port.on('data', function(data) {
	console.log(data);
});
port.on(     'error', showError);
port.open(function (err) {
	if (err) {
		return console.log('Error opening port: ', err.message);
	}
	// write errors will be emitted on the port since there is no callback to write
	sendReadStatus();
});
function showError(error) {
	console.log('Serial port error: ' + error);
}
```

techninja, 2017-06-10:
#1171 is a dupe of this (this one was first). I'm seeing this only on Windows, totally kills output and expected promise return/callback. I will look into this a bit, but _goodness_ it's layered. 😞

techninja, 2017-06-10:
I was wrong, the output killer _isn't_ this.. though this does appear to be erroring reliably.

reconbot, 2017-06-20:
Fixed in the latest two betas. Thanks for the report!

closed 2017-06-20 by reconbot