serialport/node-serialport #1162: No data event is beeing fired.

stefanpoeter, 2017-05-08:
- __SerialPort Version__: 4.0.7
- __NodeJS Version__: 6.10.3
- __Operating System__ and __Hardware Platform__: Ubuntu 16.04

- Have you checked the right __version__ of the __api docs?__: yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? no
- Are you using __Electron__ and have you checked the _Electron Docs_?: no

### Summary of Problem

I am having trouble sending or receiving data (modbus/rtu protocol). I followed the example code to write data to a serial port. The callback method on the port.write is beeing called. The debug message tells me that the data has been send. But I do not get a data event with a response.

When I use another software tool(qmodbus) to send serial data everything works fine. I checked the data that is beeing send. The qmodbus data equals my data. Baudrate and all other options match also.

### Steps and Code to Reproduce the Issue

Implementing the example from the readme.

Debug message
```
serialport write 8 bytes of data +1ms
```

HanSunyi, 2017-05-15:
@stefanpoeter I got the same problem,here is my problem
https://github.com/EmergingTechnologyAdvisors/node-serialport/issues/1169
Have you find a solution?

stefanpoeter, 2017-05-15:
Not yet.

HanSunyi, 2017-05-15:
@stefanpoeter  I found a solution,you could use a buffer to control your data
you could receive data like below

port.on('data', function(data) {
         // don't use data directlly,because it could be half-breaked.
         buffer.push.apply(buffer,data); // define a global buffer,and put data into it
         ...
         // your code here
         ...          
         //Be careful, don't foget to control the length of buffer
         // It might cause OutOfMemoryException
}

PS: I used a times counter,if can't resolve the message in buffer more than 5 times ,I reset the buffer to [],and timerCount = 0

stefanpoeter, 2017-05-15:
That is not my issue. I think my data wasn't really send through a linux socket. Reading chopped data is not the case here. This is probably a linux specific issue.

reconbot, 2017-06-20:
What's the name of the port? Do you have example code? 

I'm going to close this issue due to it's age, but if you'd like to continue with it feel free to comment and we'll reopen.

closed 2017-06-20 by reconbot