serialport/node-serialport #1170: Failed to build 5.0.0-beta* on Windows 10 x64

akozhemiakin, 2017-05-18:
- __SerialPort Version__: 5.0.0-beta3 (also 5.0.0-beta2 and 5.0.0-beta1)
- __NodeJS Version__: 7.10.0
- __Operating System__ and __Hardware Platform__: Windows 10 x64

- Have you checked the right __version__ of the __api docs?__: Yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? Yes
- Are you using __Electron__ and have you checked the _Electron Docs_?: No

### Summary of Problem
Node-pre-gyp failed to build serialport on Windows 10 x64 with the following fatal errors:
```
D:\projects\test\node_modules\nan\nan.h(24): fatal error C1083: Cannot open include file: 'node_version.h': No such file or directory (compiling source file ..\src\serialport_win.cpp) [D:
\projects\test\node_modules\serialport\build\serialport.vcxproj]
         D:\projects\test\node_modules\nan\nan.h(24): fatal error C1083: Cannot open include file: 'node_version.h': No such file or directory (compiling source file ..\src\serialport.cpp) [D:\pro
jects\test\node_modules\serialport\build\serialport.vcxproj]
```
However I have no problem with 4.0.7 installation.

reconbot, 2017-06-20:
What about node8? I'm dropping support for node7. I'm willing to be this is a node-gyp issue.

hanfengcan, 2017-06-29:
I am use Node 8 , it OK

closed 2017-06-30 by reconbot