serialport/node-serialport #1171: Uncaught (in promise) TypeError: First argument must be an int

ilya-fedin, 2017-05-19:
- __SerialPort Version__: 5.0.0-beta3
- __NodeJS Version__: 7.10.0
- __Operating System__ and __Hardware Platform__: Windows 7 x64

- Have you checked the right __version__ of the __api docs?__: Yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? Yes
- Are you using __Electron__ and have you checked the _Electron Docs_?: Yes

### Summary of Problem
Error in console:
```
Uncaught (in promise) TypeError: First argument must be an int
    at Promise (C:\Users\ilya\electron-quick-start\node_modules\serialport\lib\util.js:13)
    at C:\Users\ilya\electron-quick-start\node_modules\serialport\lib\util.js:5
    at read.then (C:\Users\ilya\electron-quick-start\node_modules\serialport\lib\bindings\win32.js:39)
```
And port response isn't read, but data are writed to port
[image]

### Steps and Code to Reproduce the Issue
I just run it in Electron Developer Tools:
```
var SerialPort = require('serialport');

var atscr = new SerialPort('COM1', function() {
	atscr.flush(function() {
		atscr.write('AT^HWVER\r', function() {
			atscr.drain(function() {
				console.log(atscr.read(4096));
			});
		});
	});
});
```

jayalfredprufrock, 2017-06-14:
FYI I was about to submit a pull request to fix this when I noticed it was already fixed in master:
https://github.com/EmergingTechnologyAdvisors/node-serialport/commit/0b2a5e30df18fdaa08465f49dfb99df23e259e69
Hopefully a beta-4 will drop soon.

ilya-fedin, 2017-06-15:
@jayalfredprufrock A lot of thanks! I will try to check tomorrow (i.e. this afternoon, at me 2:03 now)

reconbot, 2017-06-17:
Published [`serialport@5.0.0-beta4`](https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/tag/5.0.0-beta4)

reconbot, 2017-06-17:
That should solve that problem, thanks for both reporting this and trying to fix it. People like you help get betas to release. =)

closed 2017-06-17 by reconbot