serialport/node-serialport #1172: Handoff of open connection between processes via file descriptor sharing over UNIX socket

anweiss, 2017-05-19:
I asked this in the context of the johnny-five framework earlier (https://github.com/rwaldron/johnny-five/issues/1344) and was referred here, but is there a way to perform a handover of an open board serial connection from one node process to another via file descriptor sharing over UNIX sockets? Basically, I'm trying to do something like this -> https://xathrya.id/2015/12/09/nodejs-unix-sockets/ ... but passing the fd of the open serial connection to another process.

reconbot, 2017-05-19:
We expose the file descriptor as `port.fd` That might be what you want. I've never seen this done before. Sounds awesome.

anweiss, 2017-05-19:
Awesome, thanks!

anweiss, 2017-05-19:
Is there a way I can get access to the instance of the `SerialPort` object created by johnny-five upon `new Board()` instantiation?

reconbot, 2017-05-19:
yes. I'm pretty sure it's `board.port` I'd use the repl to make sure.

anweiss, 2017-05-19:
Yea, so it looks like `board.port` is a string rather than an instance of the `SerialPort` object. I'm trying to, instead, instantiate a board as follows:

```node
var port = new SerialPort("/dev/cu.usbmodem####");
var board = new five.Board({ port: port });
```

But for some reason I get a timeout error when connecting to the Arduino Uno I'm using in this case. When I don't pass in a new `SerialPort` object and just specify the device path, it connects fine.

reconbot, 2017-05-19:
Same version of serialport?

I'd explore the board.io object a bit  - take a look at the j5 source. It's
in there.

---
Francis Gulotta
wizard@roborooter.com

anweiss, 2017-05-19:
Disregard. Had the wrong baud rate. StandardFirmata is 57600. Thanks

anweiss, 2017-05-19:
So to wrap this issue up, can one pass an open file descriptor in to the instantiation of a new `SerialPort` object?

reconbot, 2017-05-19:
Not officially. I think if you create an object, don't auto open it, and
then assign the fd, It might work on version 4, it will not work with
version 5-beta however. It would be a lot easier to add to version 5.

Can I ask, why don't you pass the path instead of an open fd?

-Francis

---
Francis Gulotta
wizard@roborooter.com

anweiss, 2017-05-20:
Makes sense. Essentially I'm trying to perform a zero-downtime handoff from one Node process to another, each running in separate Docker containers and communicating by way of the UNIX socket. If this can be handled in the same manner simply by passing the path instead of the open file descriptor, then that would work too. Open to suggestions here.

anweiss, 2017-05-20:
I also tried your suggestion, but get the following error when using an open file descriptor:

```
TypeError: There's no write queue for that file descriptor (write)!
```

reconbot, 2017-05-20:
Oh that makes sense, we do some setup and teardown when we open and close ports. I'm trying to do less of that in version 5 but it's not nothing. I think we have too much state (primarily read and write queues) to do a 0 downtime handoff.

anweiss, 2017-05-20:
Ah ok. Any ideas on how one could get as close as possible to zero downtime with the current version?

reconbot, 2017-05-21:
Well, even if you could pass the FD around there will still be the chance of outstanding reads and writes. I think opening and closing is your only bet at the moment. I'll need to give this some more thought as it would be neat to accommodate passing the fd and pausing and resuming the serial stream in 5x.

anweiss, 2017-05-21:
Awesome! Yea let me know what you come up with. Will keep the issue open to track.

reconbot, 2017-06-20:
5.0.0-beta5 no longer has a write queue or state on open and close. We probably need a way to have you pass the FD into the binding but it should work once we do that.

reconbot, 2017-07-15:
I'm not going to add direct support for this but it can be accomplished by the following in `serialport@5.0.0-beta7`. (not tested)

```js
const SerialPort = require('serialport')
const port = new SerialPort('/anyPath', { autoOpen: false })
port.binding.fd = alreadOpenFd
console.log(port.isOpen) // should be true
```

This skips the open step which sets the baudrate and all the other settings. You'll probably want to pause the port and ensure no writes are going on before handing off the fd since the poller will be waiting for more data.

closed 2017-07-15 by reconbot