serialport/node-serialport #1173: Intermittent Segmentation faults

RichAyotte, 2017-05-20:
- __SerialPort Version__: v5.0.0-beta3
- __NodeJS Version__: v7.10.0
- __Operating System__ and __Hardware Platform__: Debian Sid, kernel 4.9.25, ftdi_sio ttyUSB0: FTDI USB Serial Device

- Have you checked the right __version__ of the __api docs?__: yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? no
- Are you using __Electron__ and have you checked the _Electron Docs_?: no

### Summary of Problem
I'm seeing the occasional segmentation fault.

```
[Sat May 20 10:54:33 2017] usb 6-2: new full-speed USB device number 25 using ohci-pci
[Sat May 20 10:54:33 2017] usb 6-2: New USB device found, idVendor=0403, idProduct=6001
[Sat May 20 10:54:33 2017] usb 6-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Sat May 20 10:54:33 2017] usb 6-2: Product: FT232R USB UART
[Sat May 20 10:54:33 2017] usb 6-2: Manufacturer: FTDI
[Sat May 20 10:54:33 2017] usb 6-2: SerialNumber: A9CBJ19X
[Sat May 20 10:54:33 2017] ftdi_sio 6-2:1.0: FTDI USB Serial Device converter detected
[Sat May 20 10:54:33 2017] usb 6-2: Detected FT232RL
[Sat May 20 10:54:33 2017] usb 6-2: FTDI USB Serial Device converter now attached to ttyUSB0
[Sat May 20 10:54:39 2017] usb usb6-port2: disabled by hub (EMI?), re-enabling...
[Sat May 20 10:54:39 2017] usb 6-2: USB disconnect, device number 25
[Sat May 20 10:54:39 2017] ftdi_sio ttyUSB0: FTDI USB Serial Device converter now disconnected from ttyUSB0
[Sat May 20 10:54:39 2017] ftdi_sio 6-2:1.0: device disconnected
[Sat May 20 10:54:39 2017] usb 6-2: new full-speed USB device number 26 using ohci-pci
[Sat May 20 10:54:40 2017] usb 6-2: New USB device found, idVendor=0403, idProduct=6001
[Sat May 20 10:54:40 2017] usb 6-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Sat May 20 10:54:40 2017] usb 6-2: Product: FT232R USB UART
[Sat May 20 10:54:40 2017] usb 6-2: Manufacturer: FTDI
[Sat May 20 10:54:40 2017] usb 6-2: SerialNumber: A9CBJ19X
[Sat May 20 10:54:40 2017] ftdi_sio 6-2:1.0: FTDI USB Serial Device converter detected
[Sat May 20 10:54:40 2017] usb 6-2: Detected FT232RL
[Sat May 20 10:54:40 2017] usb 6-2: FTDI USB Serial Device converter now attached to ttyUSB0
[Sat May 20 11:41:19 2017] node[17960]: segfault at 1810 ip 00007f948caeb5a1 sp 00007ffda3e334b0 error 4 in serialport.node[7f948cae0000+d000]
[Sat May 20 12:01:22 2017] node[20745]: segfault at e70 ip 00007fa2f4e395a1 sp 00007ffc3435bd20 error 4 in serialport.node[7fa2f4e2e000+d000]
[Sat May 20 12:10:40 2017] node[22869]: segfault at e70 ip 00007f8f079f25a1 sp 00007ffe150d7630 error 4 in serialport.node[7f8f079e7000+d000]
[Sat May 20 12:37:28 2017] node[23454]: segfault at 1810 ip 00007fab063185a1 sp 00007ffc6cbf6560 error 4 in serialport.node[7fab0630d000+d000]
```

### Steps and Code to Reproduce the Issue

```js
const port = new SerialPort('/dev/ttyUSB0', {
	baudRate: 9600
	, dataBits: 7
	, parity: 'even'
	, stopBits: 1
})

const parser = port.pipe(new SerialPort.parsers.ByteLength({
	length: APEX_MESSAGE.length.head
		+ APEX_MESSAGE.length.received
		+ APEX_MESSAGE.length.tail
}))

port.on('error', error => logger.error(error))

// I'm leaving this bit here for completeness but it's probably safe to ignore.
parser.on('data', message => {
	state.copy(oldState)
	messageToState(message)
	const stateDiff = getStateDiff()

	Object.keys(stateDiff).forEach(i => {
		Object.keys(APEX_MESSAGE.status[`byte${i}`]).forEach(j => {
			const bitDiff = stateDiff[i] & APEX_MESSAGE.status[`byte${i}`][j]
			if (bitDiff) {
				if (bitDiff & state[i]) {
					this.emit(j, true)
					return
				}
				this.emit(j, false)
			}
		})
	})
})

port.on('open', () => {
        // Here I have to write the port every 200ms - not sure if that's the cause.
	setInterval(() => {
		// Message needs to be re-created each time to alternate the acks
		const message = commandToMessage({
			data: [
				0x7f
				, APEX_MESSAGE.command.byte1.escrow
				, 0
			]
			, type: 'sent'
		})

		port.write(message, err => {
			if (err) {
				console.log('Error on write: ', err.message)
			}
		})
	}, 200)
	port.read()
})

```

RichAyotte, 2017-05-20:
This could be a cleanup problem so I've just added:
```js
port.on('open', () => {
	process.once('SIGUSR2', () => {
		port.close(() => process.kill(process.pid, 'SIGUSR2'))
	})
...
```
I'll test it out and report back.

RichAyotte, 2017-05-20:
It didn't help; just saw another segfault. Hopefully this will help.
```
PID 11939 received SIGSEGV for address: 0x1810
/home/rich/Projects/bitcoin-barber-btm/backend/node_modules/segfault-handler/build/Release/segfault-handler.node(+0x1a4b)[0x7f29924dea4b]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x110c0)[0x7f2994e9e0c0]
/home/rich/Projects/bitcoin-barber-btm/backend/node_modules/serialport/build/Release/serialport.node(_ZN10ReadPoller6_closeEv+0x21)[0x7f299209c5a1]
node(+0x135f090)[0x55978c1dc090]
node(uv_run+0x14c)[0x55978c1cbf0c]
node(_ZN4node5StartEP9uv_loop_siPKPKciS5_+0x6f7)[0x55978bf20e37]
node(_ZN4node5StartEiPPc+0x10d)[0x55978bf1f9ad]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf1)[0x7f2994b0f2b1]
node(_start+0x2a)[0x55978b40241a]
```

reconbot, 2017-06-20:
Please try `v5.0.0-beta5` I think I cleaned up an bug on writes.

reconbot, 2017-07-10:
Other reports of this have been solved `serialport@5.0.0-beta6` is probably going to be the last beta and doesn't have this issue. =)

closed 2017-07-10 by reconbot