serialport/node-serialport #1176: how to judge port state?

keyiis, 2017-05-26:
- __SerialPort Version__:5.0.0-beta3
- __NodeJS Version__:6.9
- __Operating System__ and __Hardware Platform__:win10 x64

- Have you checked the right __version__ of the __api docs?__:yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs?no
- Are you using __Electron__ and have you checked the _Electron Docs_?:yes

### Summary of Problem
After i replug the device,SerialPort can't acess port
```
Unhandled Promise rejection: Reading from COM port (ReadFile): Access denied ; Zone: <root> ; Task: Promise.then ; Value: Error: Reading from COM port (ReadFile): Access denied Error: Reading from COM port (ReadFile): Access denied
```
I must restart app to resolve.has some way to judge port open state by port number?

### Steps and Code to Reproduce the Issue

reconbot, 2017-06-19:
Try beta4 - it has a fix you might need.

reconbot, 2017-07-05:
Beta6 is now out it should work a lot better for this with regards to disconnect events and reconnecting.

`port.isOpen` is also available to you.

I'm going to close this issue due to it's age, but if you'd like to continue with it feel free to comment and we'll reopen.

closed 2017-07-05 by reconbot