serialport/node-serialport #1178: Can I switch parsers when reading data?

DangMin, 2017-05-30:
- __SerialPort Version__: 4.0.7
- __NodeJS Version__: 7.9.0
- __Operating System__ and __Hardware Platform__: Ubuntu 17.04

- Have you checked the right __version__ of the __api docs?__: yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs?
- Are you using __Electron__ and have you checked the _Electron Docs_?:

### Summary of Problem
Hi,

I'm doing my project which I use serialport to read and write data to the machine. Usually the messages start with [0x02] and end with [0x0D], according to the document, I should use:

`... parser: SerialPort.parsers.byteDelimiter([0x02, 0x0D]) ...`

to read the message. However, the first message when the communication start is only [0x06] as ACK signal. Therefore, I would like to ask whether it's possible to switch between parsers or not?

Thanks,

Dang Min

### Steps and Code to Reproduce the Issue
```
const port = new SerialPort(defaultPort, {
  baudRate: 9600,
  dataBits: 8,
  parity: 'none',
  stopBits: 1,
  autoOpen: false,
  parser: SerialPort.parsers.byteDelimiter([[0x02], [0x0D]])
})
```
Messages usually look like this: 
`const MsgI = [0x02, 0x30, 0x49, 0x03, 0x79, 0x0D]`

reconbot, 2017-05-30:
Nope, but it is possible to make a smarter parser than the ones we provide.
Check their source on the 4x branch.

DangMin, 2017-05-30:
Thank you for your advice. I would add a few line into the library as
```
byteDelimiter: function(delimiter) {
    if (Object.prototype.toString.call(delimiter) !== '[object Array]') {
      delimiter = [ delimiter ];
    }
    var buf = [];
    var nextDelimIndex = 0;
    return function(emitter, buffer) {
      if (buffer[0] === 0x06) {
        emitter.emit('data', buffer);
      } else {
        for (var i = 0; i < buffer.length; i++) {
          buf[buf.length] = buffer[i];
          if (buf[buf.length - 1] === delimiter[nextDelimIndex]) {
            nextDelimIndex++;
          }
          if (nextDelimIndex === delimiter.length) {
            emitter.emit('data', buf);
            buf = [];
            nextDelimIndex = 0;
          }
        }
      }
    };
```

It works for me at the moment. Hope it doesn't bother you. :+1:

closed 2017-05-30 by DangMin