serialport/node-serialport #1183: Install fails w/ Nodejs 8.x

robogeek, 2017-05-31:
- __SerialPort Version__: 4.0.7 (as dependency of node-modbus)
- __NodeJS Version__: 8.0
- __Operating System__ and __Hardware Platform__: Ubuntu

- Have you checked the right __version__ of the __api docs?__:  N/A This is about package installation
- Are you having trouble installing and you checked the __Installation Special Cases__ docs?  Yes, doesn't seem applicable
- Are you using __Electron__ and have you checked the _Electron Docs_?:  N/A

### Summary of Problem

I just installed Node.js 8.0 -- Want to install the node-modbus package, which depends on serialport@4.0.7 and that installation fails.  Fails the same way with `npm install serialport`

### Steps and Code to Reproduce the Issue

With `nvm` type:  `nvm install 8`

Then: `npm install serialport`

Among the output, this message is printed

```
node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v57-linux-x64.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and node@8.0.0 (node-v57 ABI) (falling back to source compile with node-gyp) 
node-pre-gyp http 404 status code downloading tarball https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v57-linux-x64.tar.gz 
```

And it proceeds with compiling source code but a zillion errors are printed, starting with:

```
gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
make: Entering directory '/home/david/modbus-server/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
../src/serialport.cpp: In function ‘int getIntFromObject(v8::Local<v8::Object>, std::__cxx11::string)’:
../src/serialport.cpp:90:52: warning: ‘v8::Local<v8::Int32> v8::Value::ToInt32() const’ is deprecated: Use maybe version [-Wdeprecated-declarations]
   return getValueFromObject(options, key)->ToInt32()->Int32Value();
                                                    ^
In file included from /home/david/.node-gyp/8.0.0/include/node/node.h:63:0,
                 from ../../nan/nan.h:49,
                 from ../src/./serialport.h:7,
                 from ../src/serialport.cpp:1:
/home/david/.node-gyp/8.0.0/include/node/v8.h:9336:14: note: declared here
 Local<Int32> Value::ToInt32() const {
              ^
../src/serialport.cpp: In function ‘double getDoubleFromObject(v8::Local<v8::Object>, std::__cxx11::string)’:
../src/serialport.cpp:102:53: warning: ‘v8::Local<v8::Number> v8::Value::ToNumber() const’ is deprecated: Use maybe version [-Wdeprecated-declarations]
   return getValueFromObject(options, key)->ToNumber()->NumberValue();
                                                     ^
In file included from /home/david/.node-gyp/8.0.0/include/node/node.h:63:0,
                 from ../../nan/nan.h:49,
                 from ../src/./serialport.h:7,
                 from ../src/serialport.cpp:1:
/home/david/.node-gyp/8.0.0/include/node/v8.h:9300:15: note: declared here
 Local<Number> Value::ToNumber() const {
               ^
../src/serialport.cpp: In function ‘void EIO_AfterOpen(uv_work_t*)’:
../src/serialport.cpp:174:31: warning: ‘v8::Local<v8::Int32> v8::Value::ToInt32() const’ is deprecated: Use maybe version [-Wdeprecated-declarations]
     int fd = argv[1]->ToInt32()->Int32Value();
                               ^
```

If I switch to Node.js 7.10 the install works correctly:

```
david@nuc1:~/modbus-server$ nvm use 7
Now using node v7.10.0 (npm v4.2.0)
david@nuc1:~/modbus-server$ rm -rf node_modules/
david@nuc1:~/modbus-server$ npm install node-modbus

> serialport@4.0.7 install /home/david/modbus-server/node_modules/node-modbus/node_modules/serialport
> node-pre-gyp install --fallback-to-build

[serialport] Success: "/home/david/modbus-server/node_modules/node-modbus/node_modules/serialport/build/Release/serialport.node" is installed via remote
```

fivdi, 2017-06-03:
serialport is a native module implemented in C++ and it needs to be compiled into the appropriate binary format for a target platform. The platforms supported by serialport are documented in the [Platform Support](https://github.com/EmergingTechnologyAdvisors/node-serialport#platform-support) section of the README. If the checkbox for a "Platform / Arch" and "Node.js version" is checked in the platform support table, pre-compiled binaries are available and those binaries will be installed by `npm install serialport`. Currently, there are no pre-compiled binaries for Node.js 8 so `npm install serialport` will perform the compile. Unfortunately the output is misleading in this case. For example, the following output is advertised as an error but all it means is that there are no pre-compiled binaries.
```
node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v57-linux-x64.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and node@8.0.0 (node-v57 ABI) (falling back to source compile with node-gyp) 
node-pre-gyp http 404 status code downloading tarball https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v57-linux-x64.tar.gz 
```
The compile process then outputs several warnings but no errors. Or at least there are no errors in the compiler output above. These warnings can be ignored a serialport is very likely to function correctly.

hanfengcan, 2017-06-05:
@fivdi a err is tried to download(404)。maybe the problem of net。I think you maybe need this https://npm.taobao.org/mirrors。or this http://www.jianshu.com/p/696fbdeb5b8a 

`node-pre-gyp rebuild --target=1.6.2 --arch=x64 --target_arch=x64 --dist-url=https://npm.taobao.org/mirrors/atom-shell`

infusion, 2017-06-11:
Any progress on this?

jmorris644, 2017-06-12:
If it helps, I have node 8 on a Raspberry Pi and I installed it with the --build-from-source option and it worked fine. It did have errors but gave me an "ok at the end.

infusion, 2017-06-12:
Well yeah, that's the default fallback if no pre-built package is available. Unfortunately it fails on Travis.

fivdi, 2017-06-13:
serialport never had pre-built packages for any of the ARM architectures and the build was always performed on install.

@infusion does Travis support Node.js 8? According to the [documentation](https://docs.travis-ci.com/user/languages/javascript-with-nodejs/) it supports 0.6.x through 7.7.x but not 8.x.x.

infusion, 2017-06-13:
Travis has 8.x.x support. Actually, they were really quick on providing 8.0.0 (within a few hours). At the moment it's:

```
node-pre-gyp info using node@8.1.0 | linux | x64
```

I thought there were pre-builds because of:

```
node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v57-linux-x64.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and node@8.1.0 (node-v57 ABI) (falling back to source compile with node-gyp) 
node-pre-gyp http 404 status code downloading tarball https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v57-linux-x64.tar.gz 
```

The errors in the Travis log indicate several problems with nan:

```
../../nan/nan_maybe_43_inl.h:105:8: error: ‘Maybe’ does not name a type
../../nan/nan_maybe_43_inl.h:115:8: error: ‘MaybeLocal’ does not name a type
../../nan/nan_converters.h:29:1: error: ‘return_t’ does not name a type
/home/travis/.node-gyp/8.1.0/include/node/v8.h:316:1: error: expected unqualified-id before ‘using’
/home/travis/.node-gyp/8.1.0/include/node/v8.h:6356:35: error: ‘virtual v8::EmbedderHeapTracer::~EmbedderHeapTracer()’ declared virtual cannot be defaulted in the class body
../../nan/nan_converters_43_inl.h:25:1: error: ‘return_t’ in ‘struct Nan::imp::ToFactory<v8::Object>’ does not name a type
```

and quite some more. If you want, I copy&paste the whole log here.

fivdi, 2017-06-13:
Are those compile errors not related to the version of g++ used to compile?
g++ 4.8 or higher is needed (see [here](https://docs.travis-ci.com/user/languages/javascript-with-nodejs#Node.js-v4-(or-io.js-v3)-compiler-requirements).)

infusion, 2017-06-13:
Hmm, thanks for the g++ 4.8 hint! With it, it compiles on travis now. Strange is, that it worked with node 7.
There are only deprecated-warnings remaining now.

fivdi, 2017-06-13:
>  Strange is, that it worked with node 7.

It's probably not that strange. There are pre-built binaries for v7 on the x64 so they would have been installed. This means that no compile was needed for v7 and there were therefore no compile errors. With v8 the compile is necessary as there are no pre-built binaries.

reconbot, 2017-06-17:
Our travis uses NVM we should be able to support precompiled builds for non arm platforms for node 8. There might be an issue with supporting a new gcc #1185 but I haven't looked into it yet.

PR #1194 (2017-06-17): build for node 8 and drop node 7 support
closes #1183
closed 2017-06-17 by reconbot