serialport/node-serialport #1185: g++ 6.2.0 and errno assignment issue

klugja, 2017-06-06:
- __SerialPort Version__:  (1.7.4)
- __NodeJS Version__: 0.10.48
- __Operating System__ and __Hardware Platform__:Linux/openembedded/Arm

We are limited by the instruction set of our CPU to 0.10.48 NodeJS.

- Have you checked the right __version__ of the __api docs?__:Not relevent
- Are you having trouble installing and you checked the __Installation Special Cases__ docs?No
- Are you using __Electron__ and have you checked the _Electron Docs_?:No

### Summary of Problem
  CXX(target) Release/obj.target/serialport/src/serialport.o
  CXX(target) Release/obj.target/serialport/src/serialport_unix.o
  CXX(target) Release/obj.target/serialport/src/serialport_poller.o
../src/serialport_poller.cpp: In member function 'void SerialportPoller::callCallback(int)':
../src/serialport_poller.cpp:38:35: error: invalid pure specifier (only '= 0' is allowed) before '(' token
     uv_err_t errno = uv_last_error(uv_default_loop());
                                   ^
../src/serialport_poller.cpp:38:53: error: function 'uv_err_t* __errno_location()' is initialized like a variable
     uv_err_t errno = uv_last_error(uv_default_loop());
                                                     ^

### Steps and Code to Reproduce the Issue
Build serial port code with uv_err_t errno = and g++ 6.2

klugja, 2017-06-06:
diff -Naru old/node_modules/node-red/node_modules/node-red-node-serialport/node_modules/serialport/src/serialport_poller.cpp new/node_modules/node-red/node_modules/node-red-node-serialport/node_modules/serialport/src/serialport_poller.cpp
--- old/node_modules/node-red/node_modules/node-red-node-serialport/node_modules/serialport/src/serialport_poller.cpp	2015-05-09 10:20:06.000000000 -0500
+++ new/node_modules/node-red/node_modules/node-red-node-serialport/node_modules/serialport/src/serialport_poller.cpp	2017-06-06 13:19:56.156266989 -0500
@@ -35,8 +35,8 @@
     #ifdef UV_ERRNO_H_
     const char* err_string = uv_strerror(status);
     #else
-    uv_err_t errno = uv_last_error(uv_default_loop());
-    const char* err_string = uv_strerror(errno);
+    uv_err_t err = uv_last_error(uv_default_loop());
+    const char* err_string = uv_strerror(err);
     #endif
     snprintf(this->errorString, sizeof(this->errorString), "Error %s on polling", err_string);
     argv[0] = v8::Exception::Error(NanNew<v8::String>(this->errorString));

klugja, 2017-06-06:
`diff -Naru old/node_modules/node-red/node_modules/node-red-node-serialport/node_modules/serialport/src/serialport_poller.cpp new/node_modules/node-red/node_modules/node-red-node-serialport/node_modules/serialport/src/serialport_poller.cpp
--- old/node_modules/node-red/node_modules/node-red-node-serialport/node_modules/serialport/src/serialport_poller.cpp	2015-05-09 10:20:06.000000000 -0500
+++ new/node_modules/node-red/node_modules/node-red-node-serialport/node_modules/serialport/src/serialport_poller.cpp	2017-06-06 13:19:56.156266989 -0500
@@ -35,8 +35,8 @@
     #ifdef UV_ERRNO_H_
     const char* err_string = uv_strerror(status);
     #else
-    uv_err_t errno = uv_last_error(uv_default_loop());
-    const char* err_string = uv_strerror(errno);
+    uv_err_t err = uv_last_error(uv_default_loop());
+    const char* err_string = uv_strerror(err);
     #endif
     snprintf(this->errorString, sizeof(this->errorString), "Error %s on polling", err_string);
     argv[0] = v8::Exception::Error(NanNew<v8::String>(this->errorString));
`

klugja, 2017-06-06:
```diff -Naru old/node_modules/node-red/node_modules/node-red-node-serialport/node_modules/serialport/src/serialport_poller.cpp new/node_modules/node-red/node_modules/node-red-node-serialport/node_modules/serialport/src/serialport_poller.cpp
--- old/node_modules/node-red/node_modules/node-red-node-serialport/node_modules/serialport/src/serialport_poller.cpp	2015-05-09 10:20:06.000000000 -0500
+++ new/node_modules/node-red/node_modules/node-red-node-serialport/node_modules/serialport/src/serialport_poller.cpp	2017-06-06 13:19:56.156266989 -0500
@@ -35,8 +35,8 @@
     #ifdef UV_ERRNO_H_
     const char* err_string = uv_strerror(status);
     #else
-    uv_err_t errno = uv_last_error(uv_default_loop());
-    const char* err_string = uv_strerror(errno);
+    uv_err_t err = uv_last_error(uv_default_loop());
+    const char* err_string = uv_strerror(err);
     #endif
     snprintf(this->errorString, sizeof(this->errorString), "Error %s on polling", err_string);
     argv[0] = v8::Exception::Error(NanNew<v8::String>(this->errorString));
```

reconbot, 2017-06-17:
Hi @klugja I'm sorry we're no longer support serialport 1.7.4 or node 0.10. There are some pretty hard to work with bugs in that version of serialport. I don't recall when we dropped node 0.10 support but I'd try to use version serialport-3x as that has the majority of the bug fixes and still supports most old versions of node.

Best of luck.

klugja, 2017-06-18:
You should not have closed this because you use errno across all levels and
your build is broken with g++ 6.2 across all levels.  You changed the file
name but overload errno the same way. 6.2 is the default for yocto 2.2.
Have you tested it?

reconbot, 2017-06-19:
I'll keep this open for g++ 6.2 support, I closed it because we're not going to be able to backport any changes to the versions in your original request.

reconbot, 2017-07-06:
The poller has been removed in 5.0.0-beta6, so this should no longer be an issue.

reconbot, 2017-07-06:
Please test and open a new issue if you're still having problems, I don't have a machine with g++ 6.2 so I can't test, but I don't see a reason why we can't support it.

closed 2017-07-06 by reconbot