serialport/node-serialport #1187: Memory Leaks Detected when creating & destroying large volume of connections.

adedomin, 2017-06-09:
- __SerialPort Version__: 4.0.7
- __NodeJS Version__: 4.6.0, 4.8.2, 6.10.3
- __Operating System__ and __Hardware Platform__: Debian Jessielite (armv8 - RaspPi 3 - nodesource v4), Debian Sid (x86_64), Fedora 25 (x86_64)

- Have you checked the right __version__ of the __api docs?__: yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs?: no, n/a
- Are you using __Electron__ and have you checked the _Electron Docs_?: no, n/a

### Summary of Problem
opening and closing many connections causes memory leaks.

### Steps and Code to Reproduce the Issue

any large number of opens and closes seems to cause a massive memory leak over a period of time

below is the most extreme example I could make; I used a pseudo terminal for testing on the x86 platforms.
you can set up a test using `socat -d -d pty,raw,echo=0 pty,raw,echo=0` and then consuming one of the pty's it allocates with `cat /dev/pts/something` and then using the other as the path parameter for the SerialPort object constructor.

```
var SerialPort = require('serialport'),
    port

var forever = () => {
    port = new SerialPort('/dev/pts/something', { baudRate: 38400 })
    port.on('open', (err) => {
        if (err) {
            console.log(err)
            return setImmediate(() => forever())
        }
        port.write('yyyyyyyyyyyyyyy\n', (err) => { 
            if (err) console.log(err)
            port.close((err) => {
                if (err) console.log(err)
                setImmediate(() => forever())
            }) 
        })
    })
}
forever()
```

More information forthcoming...

reconbot, 2017-06-10:
Always good stuff to know about. I'd appreciate testing on 5x beta as well.
However there are no known leaks in either.

---
Francis Gulotta
wizard@roborooter.com

adedomin, 2017-06-10:
What i can say right now is, I have a raspberry pi device in the field and when someone disconnects the serial (USB) cable while configured a certain way, it constantly recreates SerialPort Objects which results a slow increase of memory that can't be reclaimed; this ultimately results in some form of out of memory error in about a week of time.

I decided to explore the memory leak in detail, using tools like heapdump--findable on npm.
What I found was an increase in V8's resident set but not the javascript heap size portion. most of the unreclaimed memory isn't from javascript as far as I can see, as new/old pieces don't account for the massive increase in the resident set size.

I'm going to try further debugging and let you know what i find.

closed 2017-06-12 by adedomin

adedomin, 2017-06-14:
other note, this example code doesn't appear to work on the serialport@5.0.0-beta3. so that's good news.