serialport/node-serialport #1190: SerialPort is not a constructor

jmorris644, 2017-06-12:
- __SerialPort Version__:4.0.7 and 5.0.0-Beta3
- __NodeJS Version__: 8
- __Operating System__ and __Hardware Platform__: Jessie on Pi and Win10

- Have you checked the right __version__ of the __api docs?__: Yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? No prob installing
- Are you using __Electron__ and have you checked the _Electron Docs_?: n/a

### Summary of Problem

All of my testing has been using --build-from-source on Jessie and Win10 using both 4.0.7 and 5.0.0-beta3. I am getting the same error on both platforms with both serialport releases.

I am consistently getting this error across operating systems and releases of serialport.

var SerialPort = require('serialport');
var port = new SerialPort('com1:', {
                  ^
    baudRate: 57600
});

TypeError: SerialPort is not a constructor.

### Steps and Code to Reproduce the Issue

While I am waiting for hopefully an answer, I am going to try Node ver 7. Thanks Joe

reconbot, 2017-06-17:
Can you console log `SerialPort` to see what you're getting? It is most definitely a constructor. Node 4 and 6 are both tested and work.

reconbot, 2017-07-05:
I'm going to close this issue due to it's age, but if you'd like to continue with it feel free to comment and we'll reopen.

closed 2017-07-05 by reconbot

nemr, 2017-08-03:
Im getting the same error :
TypeError: SerialPort is not a constructor

Im using serialport @5.0.0 with node 6.11.1 on Raspberry pi model 3 : ARMv7 Processor rev 4 (v7l)

my Code :
var SerialPort = require('serialport');
var port = new SerialPort('/dev/ttyS0');

port.on('open', function() {
  port.write('main screen turn on', function(err) {
    if (err) {
      return console.log('Error on write: ', err.message);
    }
    console.log('message written');
  });
});

// open errors will be emitted as an error event
port.on('error', function(err) {
  console.log('Error: ', err.message);
})

reconbot, 2017-08-03:
Please open a new issue and console log what the require returns