serialport/node-serialport #1193: error when pull out device

kuangyunsheng, 2017-06-14:
- __SerialPort Version__: 4.0.7
- __NodeJS Version__:v7.3.0
- __Operating System__ and __Hardware Platform__: win7_x64

- Have you checked the right __version__ of the __api docs?__: yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? yes
- Are you using __Electron__ and have you checked the _Electron Docs_?: no

### Summary of Problem
After open a serial port and listen data event, it's ok. But, when I pull out the device from the port, error occured and the program exit auto.

### Steps and Code to Reproduce the Issue
step1, insert a gps device with serialport

step 2, open serial port via serialport-term:
serialport-term -p COM5 -b 4800

step 3:  pull out the gps device. error occured:
[image]

when I connect the gps device in my own program, this error will break off the program. 
how to avoid this error or where to catch it?

hanfengcan, 2017-06-29:
I have similar problems when pull out CH340 https://github.com/EmergingTechnologyAdvisors/node-serialport/issues/1145

reconbot, 2017-07-05:
Please try `5.0.0-beta6` this should have better disconnect support.

- disconnect events don't fire when you're closing the port
- Close during writes causes an error, close during reads does not

closed 2017-07-21 by reconbot