serialport/node-serialport #1206: prebuilt windows binaries for node 8

MikeKovarik, 2017-06-21:
- __SerialPort Version__: 4.x or 5.x
- __NodeJS Version__: 8.1.2
- __Operating System__ and __Hardware Platform__: Windows 10 64b

### Summary of Problem
Will there be prebuilt binaries for Node 8 for Windows please? I've been trying to build for two hours now and I'm slowly loosing my sanity with the whole node-gyp and million other dependencies. Thank you

reconbot, 2017-06-21:
There will be for the next beta of 5x

---
Francis Gulotta
wizard@roborooter.com

cinderblock, 2017-06-28:
What about SerialPort version 4?

Is there a list somewhere of the available pre-built binaries versions?

reconbot, 2017-06-29:
The github releases.
https://github.com/EmergingTechnologyAdvisors/node-serialport/releases

It's a lot of work to get the 4x branch build server working again. I don't
have a windows machine to develop on so I currently can't even do it by
hand.

---
Francis Gulotta
wizard@roborooter.com

reconbot, 2017-06-29:
@mattclaw did a node8 build for me of 4.0.7, will check it out and add it if it's in good shape

reconbot, 2017-07-01:
Added `serialport-v4.0.7-node-v57-win32-x64.tar.gz` give it a go!

closed 2017-07-05 by reconbot